libimobiledevice/libusbmuxd #25: (not sure it's a bug, but...) Python-Client (tcprelay.py) missing

wyatt8740, 2015-02-17:
Whatever happened to the python client? Is there a replacement for tcprelay? I used that script HEAVILY. It was incredibly useful!

FunkyM, 2015-09-25:
It was removed due to nobody maintaining it. :)
The iproxy tool should be used instead of the python examples.
Furthermore even iproxy will be superseeded by a new tool in libimobiledevice in the future.

closed 2015-09-25 by FunkyM

wyatt8740, 2016-01-13:
alright, so the functionality is still present. What a relief.

ishtiaq156, 2017-09-22:
@wyatt8740 can you share the release of libusbmuxd with python-client?

wyatt8740, 2017-10-07:
@ishtiaq156 It's still in the git history... Just checkout a commit from before it was removed. It still works. Alternately tools/iproxy still works for the same purpose in the current releases (though that's a C program). 

    $ git clone https://github.com/libimobiledevice/libusbmuxd.git
    $ cd libusbmuxd
    $ git checkout e866c7136fac5b6a915cb47e3407fb802aa1cba4