libimobiledevice/libusbmuxd #26: Device removed notification arrived when connecting another iPhone

skortela, 2015-04-16:
I have registered to listen device added/removed notifications.
Use case:
Connect iPhone, pair and trust, wait for device is ready.
Connect another iPhone, pair and trust.

When pairing another iPhone, i got notification device removed notification from BOTH devices.
after a while both devices will arrive again.

nikias, 2015-04-16:
Please understand that the device will reconnect via USB when you tap "Trust". This is the intended behavior. I am pretty sure this is what you observe here.

skortela, 2015-04-16:
I understand that, device is reconnecting, but I have two devices connected
and both are reconnected when trusting the later device.
Should I get reconnect only from later device?

nikias, 2015-04-17:
Hmm indeed you should only get the reconnect for the second device if the first one already reconnected. Also I read you only have this on Mac but not on Windows which is an inconsistent behavior. I will check if I see this on OSX but I don't remember this happening for me.

skortela, 2015-06-23:
Any update on this?

nikias, 2017-01-02:
@skortela is this still happening for you? I never really followed up on this.

skortela, 2017-01-02:
@nikias Could not reproduce.
Now there comes IDEVICE_DEVICE_PAIRED notification once device is paired.
IDEVICE_DEVICE_REMOVE arrives only when USB-cable is removed.
Can be closed.

closed 2017-01-02 by skortela

nikias, 2017-01-02:
@skortela thanks