libimobiledevice/libusbmuxd #31: Compiling libusbmuxd on windows

hyunmin90, 2015-07-22:
So I have successfully got to a point where I am compiling libusbmuxd on windows mingw to use it for windows platform.

However although autogen.sh and configure is done without error 

it fails to make libusbmuxd at below . Does anyone have any idea on this? Is there a clear instruction on how to build in windows? Quoatmotion seems to have something but they don't detail anything on how to compile...

$ make
make  all-recursive
make[1]: Entering directory `/home/Naver/libusbmuxd-1.0.10'
Making all in common
make[2]: Entering directory `/home/Naver/libusbmuxd-1.0.10/common'
  CC     collection.lo
  CCLD   libinternalcommon.la
make[2]: Leaving directory `/home/Naver/libusbmuxd-1.0.10/common'
Making all in src
make[2]: Entering directory `/home/Naver/libusbmuxd-1.0.10/src'
  CC     libusbmuxd.lo
libusbmuxd.c:46:26: error: expected ';', ',' or ')' before numeric constant
 #define sleep(x) Sleep(x*1000)
                          ^
libusbmuxd.c:46:25: error: expected ';', ',' or ')' before '*' token
 #define sleep(x) Sleep(x*1000)
                         ^
make[2]: *** [libusbmuxd.lo] Error 1
make[2]: Leaving directory `/home/Naver/libusbmuxd-1.0.10/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/Naver/libusbmuxd-1.0.10'
make: *** [all] Error 2

aburgh, 2015-07-22:
Is that line of source really `Sleep(x1000)`? The repo is `Sleep(x*1000)`: https://github.com/libimobiledevice/libusbmuxd/blob/master/src/libusbmuxd.c#L46. I think the Markdown processing swallowed the asterisk. You can use back ticks to post text verbatim (see the Markdown link above the text box when posting).

hyunmin90, 2015-07-23:
So apparently, compiling in mingw causes a problem with the libusbmuxd on line for sleep.
For those who suffer the same issue as me, have a go and delete the line. 

IT worked fine without the sleep define and now it works flawlessly.

wizcpp, 2016-01-24:
$ make
make  all-recursive
make[1]: Entering directory `/home/AOE/libusbmuxd-master'
Making all in common
make[2]: Entering directory `/home/AOE/libusbmuxd-master/common'
make[2]: Nothing to be done for `all'.
make[2]: Leaving directory `/home/AOE/libusbmuxd-master/common'
I have met the save error prompts.
----------

Making all in src
make[2]: Entering directory `/home/AOE/libusbmuxd-master/src'
  CC     libusbmuxd.lo
libusbmuxd.c:46:26: error: expected ';', ',' or ')' before numeric constant
 #define sleep(x) Sleep(x*1000)
                          ^
libusbmuxd.c:46:25: error: expected ';', ',' or ')' before '*' token
 #define sleep(x) Sleep(x*1000)
                         ^
make[2]: *** [libusbmuxd.lo] Error 1
make[2]: Leaving directory `/home/AOE/libusbmuxd-master/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/AOE/libusbmuxd-master'
make: *** [all] Error 2

sesjrs, 2016-02-12:
Interchanging the #include directives will be needed.

In the libusbmuxd.c file,
move #include <unistd.h> line before #ifdef WIN32.

daniel-bluesea, 2016-12-05:
@hyunmin90 @sesjrs both your solutions solve the problem.
i think you can close this issue

daniel-bluesea, 2016-12-13:
@FunkyM @nikias what is the preferred solution (delete the Sleep line or move  #include <unistd.h> line before #ifdef WIN32.) according to you?

nikias, 2017-01-02:
Can you guys try my patch here: https://gist.github.com/nikias/1822406ace368657a35703f68000be6c
it *should* make it compile properly when the sleep() function is available and #define it when not.

daniel-bluesea, 2017-01-03:
seems ok.
thanks

nikias, 2017-01-03:
Should be fixed with commit https://github.com/libimobiledevice/libusbmuxd/commit/5b338a411203cd42e7b09d56585e1acb408f406e

closed 2017-01-03 by nikias