libimobiledevice/libusbmuxd #34: Cannot compile on windows with msys2

luckcolors, 2015-08-12:
Hello i'm trying to compile libusbmuxd but there's a linker problem that i don't know how to solve.
Here's the full log i also included libplist compile log so you have more info:

Note: it seems that libusbmuxd is not able to detect libplist automatically so i had to tell where it was placed.

luckc@DESKTOP-CG79PC9 MSYS ~/libimobiledevice/libusbmuxd
$ libplist_LIBS=/usr/local/lib libplist_CFLAGS=/usr/local/include ./autogen.sh
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:28: installing './compile'
configure.ac:6: installing './missing'
common/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-pc-msys
checking host system type... x86_64-pc-msys
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/x86_64-pc-msys/bin/ld.exe
checking if the linker (/usr/x86_64-pc-msys/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-pc-msys file names to x86_64-pc-msys format... func_convert_file_noop
checking how to convert x86_64-pc-msys file names to toolchain format... func_convert_file_noop
checking for /usr/x86_64-pc-msys/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
[... 432 lines cut ...]
make[2]: ingresso nella directory "/home/luckc/libimobiledevice/libplist/tools"
  CC       plistutil.o
  CCLD     plistutil.exe
make[2]: uscita dalla directory "/home/luckc/libimobiledevice/libplist/tools"
Making all in cython
make[2]: ingresso nella directory "/home/luckc/libimobiledevice/libplist/cython"
/usr/bin/cython -I. -I../src -o plist.c plist.pyx
make  all-am
make[3]: ingresso nella directory "/home/luckc/libimobiledevice/libplist/cython"
  CC       plist_la-plist_util.lo
  CC       plist_la-plist.lo
  CCLD     plist.la
libtool: warning: undefined symbols not allowed in x86_64-pc-msys shared libraries; building static only
make[3]: uscita dalla directory "/home/luckc/libimobiledevice/libplist/cython"
make[2]: uscita dalla directory "/home/luckc/libimobiledevice/libplist/cython"
Making all in test
make[2]: ingresso nella directory "/home/luckc/libimobiledevice/libplist/test"
  CC       plist_cmp.o
  CCLD     plist_cmp.exe
  CC       plist_test.o
  CCLD     plist_test.exe
make[2]: uscita dalla directory "/home/luckc/libimobiledevice/libplist/test"
make[2]: ingresso nella directory "/home/luckc/libimobiledevice/libplist"
make[2]: uscita dalla directory "/home/luckc/libimobiledevice/libplist"
make[1]: uscita dalla directory "/home/luckc/libimobiledevice/libplist"

aburgh, 2015-08-13:
Did you also install libplist with `make install`? If that isn't it, try building libusbmuxd again with the command `make V=1` to get verbose output that will show the full command for the compiler invocation.

Also, when you post command output to Github, you can use a line with three ticks before and after the quoted text to preserve all the text and formatting, making it easier to read. For example, from https://guides.github.com/features/mastering-markdown/#examples:

```
    ```
    if (isAwesome){
      return true
    }
    ```
```

luckcolors, 2015-08-13:
Ok recompiled the package with the verbose option. Also there's a warning that sais that the lib is not compiled shared but static.

```
luckc@DESKTOP-CG79PC9 MSYS ~/libimobiledevice/libplist
$ make V=1
make  all-recursive
make[1]: ingresso nella directory "/home/luckc/libimobiledevice/libplist"
Making all in libcnary
make[2]: ingresso nella directory "/home/luckc/libimobiledevice/libplist/libcnary"
/bin/sh ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node.lo -MD -MP -MF .deps/node.Tpo -c -o node.lo node.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node.lo -MD -MP -MF .deps/node.Tpo -c node.c  -DDLL_EXPORT -DPIC -o .libs/node.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node.lo -MD -MP -MF .deps/node.Tpo -c node.c -o node.o >/dev/null 2>&1
mv -f .deps/node.Tpo .deps/node.Plo
/bin/sh ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT list.lo -MD -MP -MF .deps/list.Tpo -c -o list.lo list.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT list.lo -MD -MP -MF .deps/list.Tpo -c list.c  -DDLL_EXPORT -DPIC -o .libs/list.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT list.lo -MD -MP -MF .deps/list.Tpo -c list.c -o list.o >/dev/null 2>&1
mv -f .deps/list.Tpo .deps/list.Plo
/bin/sh ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node_list.lo -MD -MP -MF .deps/node_list.Tpo -c -o node_list.lo node_list.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node_list.lo -MD -MP -MF .deps/node_list.Tpo -c node_list.c  -DDLL_EXPORT -DPIC -o .libs/node_list.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node_list.lo -MD -MP -MF .deps/node_list.Tpo -c node_list.c -o node_list.o >/dev/null 2>&1
mv -f .deps/node_list.Tpo .deps/node_list.Plo
/bin/sh ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT iterator.lo -MD -MP -MF .deps/iterator.Tpo -c -o iterator.lo iterator.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT iterator.lo -MD -MP -MF .deps/iterator.Tpo -c iterator.c  -DDLL_EXPORT -DPIC -o .libs/iterator.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT iterator.lo -MD -MP -MF .deps/iterator.Tpo -c iterator.c -o iterator.o >/dev/null 2>&1
mv -f .deps/iterator.Tpo .deps/iterator.Plo
/bin/sh ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node_iterator.lo -MD -MP -MF .deps/node_iterator.Tpo -c -o node_iterator.lo node_iterator.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node_iterator.lo -MD -MP -MF .deps/node_iterator.Tpo -c node_iterator.c  -DDLL_EXPORT -DPIC -o .libs/node_iterator.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../libcnary/include -g -O2 -MT node_iterator.lo -MD -MP -MF .deps/node_iterator.Tpo -c node_iterator.c -o node_iterator.o >/dev [... cut ...]
libtool: link: gcc -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden -I../include -I../libcnary/include -g -O2 -o .libs/plist_test.exe plist_test.o  ../src/.libs/libplist.dll.a -lxml2 -lpthread -lz -llzma -liconv -L/usr/local/lib
make[2]: uscita dalla directory "/home/luckc/libimobiledevice/libplist/test"
make[2]: ingresso nella directory "/home/luckc/libimobiledevice/libplist"
make[2]: uscita dalla directory "/home/luckc/libimobiledevice/libplist"
make[1]: uscita dalla directory "/home/luckc/libimobiledevice/libplist"
 ```

aburgh, 2015-08-16:
Thanks for the log and tagging it with Markdown. Can you confirm that libplist is installed, i.e., that you ran `make install` and that the static library, at least, is in `/usr/local/lib`? 

Can you also recompile libusbmuxd with `V=1`? All of the error messages report a missing symbol from libplist, so I'm looking to confirm that the library is installed in a location that the libusbmuxd searches. In particular, we want to see the details behind this step: ` CCLD     libusbmuxd.la`.

Regarding the fact that libplist only built as static, I'm not sure about GCC's ld on Windows, but the linkers I'm familiar with will fall back to using a static library automatically (I cross-compile for Windows using GCC on OS X and Mingw-w64).

luckcolors, 2015-08-22:
Ok so about libplist being installed:
```
$ ls /usr/local/lib/
libplist.a      libplist.la   libplist++.dll.a  pkgconfig
libplist.dll.a  libplist++.a  libplist++.la     python2.7
```
And here's libusbmuxd verbose log.
```
$ libplist_CFLAGS=/usr/local/lib libplist_LIBS=/usr/local/lib ./autogen.sh
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:28: installing './compile'
configure.ac:6: installing './missing'
common/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-pc-msys
checking host system type... x86_64-pc-msys
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/x86_64-pc-msys/bin/ld.exe
checking if the linker (/usr/x86_64-pc-msys/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-pc-msys file names to x86_64-pc-msys format... fu                                                                                                               nc_convert_file_noop
checking how to convert x86_64-pc-msys file names to toolchain format... func_co                                                                                                               nvert_file_noop
checking for /usr/x86_64-pc-msys/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|                                                                                                               ^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_imp                                                                                                               lib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
[... 195 lines cut ...]
.libs/libusbmuxd.o: In function `send_read_buid_packet':
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:471: undefined reference to `plist_free'
.libs/libusbmuxd.o: In function `usbmuxd_read_buid':
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1147: undefined reference to `plist_free'
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1139: undefined reference to `plist_dict_get_item'
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1140: undefined reference to `plist_get_node_type'
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1141: undefined reference to `plist_get_string_val'
.libs/libusbmuxd.o: In function `usbmuxd_read_pair_record':
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1195: undefined reference to `plist_free'
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1183: undefined reference to `plist_dict_get_item'
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1184: undefined reference to `plist_get_node_type'
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1186: undefined reference to `plist_get_data_val'
.libs/libusbmuxd.o: In function `usbmuxd_save_pair_record':
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1222: undefined reference to `plist_new_data'
/home/luckc/libimobiledevice/libusbmuxd/src/libusbmuxd.c:1235: undefined reference to `plist_free'
collect2: error: ld returned 1 exit status
Makefile:419: set di istruzioni per l'obiettivo "libusbmuxd.la" non riuscito
make[2]: *** [libusbmuxd.la] Errore 1
make[2]: uscita dalla directory "/home/luckc/libimobiledevice/libusbmuxd/src"
Makefile:465: set di istruzioni per l'obiettivo "all-recursive" non riuscito
make[1]: *** [all-recursive] Errore 1
make[1]: uscita dalla directory "/home/luckc/libimobiledevice/libusbmuxd"
Makefile:374: set di istruzioni per l'obiettivo "all" non riuscito
make: *** [all] Errore 2
```

mexmer, 2015-08-22:
not sure what distro you using, but you usually need devel package installed, because it contains headers needed for compilation.

usualy it's called libsomething-dev (at least on debian and slackware)

aburgh, 2015-08-22:
I think you have `libplist_FLAGS` and `libplist_LIBS` set wrong. Does my answer in this question help? https://github.com/libimobiledevice/libusbmuxd/issues/24

nikias, 2017-01-02:
Closing due to inactivity. Compilation works as long as the build environment and the dependencies are fine.

closed 2017-01-02 by nikias