libimobiledevice/libusbmuxd #36: /usr/sbin/asr was terminated by signal 11

Itbnolehurr, 2015-09-08:
i am trying to bypass an icloud lock for a friend and idevicrestore.exe keeps stopping at a specific point and tells me that /usr/sbin/asr was terminated by signal 11. anyway to fix this?

minhlmo, 2015-10-08:
nts = "13994007-DD55-C990-624E-E6C0175404BE"}
        MinimumSystemPartition         => <CFNumber 0x176284a0 [0xa64468]>{value
 = +2033, type = kCFNumberSInt64Type}
        SystemPartitionSize            => <CFNumber 0x17628120 [0xa64468]>{value
 = +2033, type = kCFNumberSInt64Type}
        PersonalizedDuringPreflight    => <CFBoolean 0xa6474c [0xa64468]>{value
= true}
        SystemPartitionPadding         => <CFBasicHash 0x17628280 [0xa64468]>{ty
pe = mutable dict, count = 9,
entries =>
        0 : <CFString 0x17627db0 [0xa64468]>{contents = "512"} = <CFNumber 0x176
27f40 [0xa64468]>{value = +1280, type = kCFNumberSInt64Type}
        2 : <CFString 0x176284f0 [0xa64468]>{contents = "128"} = <CFNumber 0x176
27af0 [0xa64468]>{value = +1280, type = kCFNumberSInt64Type}
        3 : <CFString 0x17627b00 [0xa64468]>{contents = "16"} = <CFNumber 0x1762
8170 [0xa64468]>{value = +160, type = kCFNumberSInt64Type}
        4 : <CFString 0x17627f80 [0xa64468]>{contents = "32"} = <CFNumber 0x1762
7da0 [0xa64468]>{value = +320, type = kCFNumberSInt64Type}
        5 : <CFString 0x17627de0 [0xa64468]>{contents = "768"} = <CFNumber 0x176
27df0 [0xa64468]>{value = +1280, type = kCFNumberSInt64Type}
        6 : <CFString 0xa37348 [0xa64468]>{contents = "8"} = <CFNumber 0x17627c9
0 [0xa64468]>{value = +80, type = kCFNumberSInt64Type}
        7 : <CFString 0x17628590 [0xa64468]>{contents = "1024"} = <CFNumber 0x17
6284d0 [0xa64468]>{value = +1280, type = kCFNumberSInt64Type}
        8 : <CFStri...
entering load_sep_os
device has no sep
entering ramrod_ticket_update
looking up root ticket hash
crypto-hash-method not found, defaulting to SHA1
device tree ticket_hash: 342423C823185136760278538023C72F2F5E2A5B
computed ticket_hash   : 342423C823185136760278538023C72F2F5E2A5B
received valid ticket (2732 bytes)
entering partition_nand_device
device supports boot-from-NAND
nand device is already partitioned
entering wait_for_storage_device
Searching for NAND service
Found NAND service: IOFlashStoragePartition
NAND initialized. Waiting for devnode.
entering clear_remap_variable
executing /usr/sbin/nvram
entering ramrod_reprobe_device_path
entering ramrod_probe_media
device partitioning scheme is GPT
find_filesystem_partitions: storage=/dev/disk0s1 system=/dev/disk0s1s1 data=/dev
/disk0s1s2 baseband data= log=
entering ramrod_reprobe_device_path
unable to stat block device /dev/disk1: No such file or directory
/dev/disk1 does not exist, skipping
entering ramrod_probe_media
device partitioning scheme is GPT
find_filesystem_partitions: storage=/dev/disk0s1 system=/dev/disk0s1s1 data=/dev
/disk0s1s2 baseband data= log=
entering ramrod_probe_media
device partitioning scheme is GPT
find_filesystem_partitions: storage=/dev/disk0s1 system=/dev/disk0s1s1 data=/dev
/disk0s1s2 baseband data= log=
executing /sbin/fsck_hfs
journal_replay(/dev/disk0s1s1) returned 0
** /dev/rdisk0s1s1
        Cache size should be multiple of cache block size (currently 32K)
        Using cacheBlockSize=32K cacheTotalBlock=2019 cacheSize=64608K.
   Executing fsck_hfs (version hfs-305.1.1).
** Checking Journaled HFS Plus volume.
** Detected a case-sensitive volume.
   The volume name is System
** Checking extents overflow file.
** Checking catalog file.
** Checking multi-linked files.
[... 205 lines cut ...]
entering restore_images
Successfully marked device node /dev/rdisk0s1s1 as static data
executing /usr/sbin/asr -source asr://localhost:12345 -target /dev/disk0s1s1 -er
ase -noprompt --chunkchecksum --puppetstrings
SYSLOG: Oct  8 05:38:42 localhost kernel[0] <Notice>: void AppleUSBDeviceMux::ha
ndleConnectResult(struct BulkUSBMuxSession *, errno_t) new session to port 12345
 failed: 61
SYSLOG: Oct  8 05:38:42 localhost asr[28] <Error>: bootstrap_look_up2 failed wit
h 0x44e
ASR: asr: Unable to disable idle sleep - -536870199
ASR STATUS: start       242.6   multicast-client
ASR: Waiting for connection attempt from server
SYSLOG: Oct  8 05:38:42 localhost asr[28] <Error>: __net_helper_get_connection_b
lock_invoke_3 could not connect to networkd
ASR STATUS: setup
ASR: Validating target...
ASR: done
ASR STATUS: metadata
ASR: Validating source...
ASR: (null)
/usr/sbin/asr was terminated by signal 11

ERROR: Unable to restore device

C:\libi>

minhlmo, 2015-10-08:
ios 9.0.1 are dark. I tried the 6.1.3, dark do not ram dicsk and BuildManifest.plist and renamed to load mode. sorry my english is not good
mail: minhmlon@gmail.com

closed 2016-10-10 by nikias