libimobiledevice/libusbmuxd #43: Crash when usbmuxd_subscribe called more than once

qmfrederik, 2016-11-13:
If you call `usbmuxd_subscribe` more than once, but do not call `usbmuxd_unsubscribe` in between, the `device_monitor` thread which is created by `usbmuxd_subscribe` is started multiple times.

When messages such as `MESSAGE_DEVICE_REMOVE` are received, _all_ of these threads will attempt to find the device, and remove the handle for that device. This leads to race conditions and ultimately to `free(devinfo)` being called twice.

I see three options to resolve this:

1. Make `usbmuxd_subscribe` check whether another callback is already registered, and if so, return `-EALREADY` to let the caller know this was not a valid operation
2. The same as above, but silently call `usbmuxd_unsubscribe` and register the new callback
3. Support handling multiple callbacks.

My preference currently goes to 1, and I can submit a PR for this, but before I do that, I'd like to check with you whether this is the project's preferred option.

jackczhang, 2016-11-30:
I have encountered this problem, too.
I preferred option 3, which can support handling multiple callbacks, because i want to do that in one process with multiple threads, but now crashed.

nikias, 2019-05-25:
Fixed with commit a6b542b389d0536d2730c1721164a712ec2f020e. New API:
```
usbmuxd_events_subscribe
usbmuxd_events_unsubscribe
```
enjoy

closed 2019-05-25 by nikias