libimobiledevice/libusbmuxd #44: ffffffffffffffffffffffffffffffffffffffff

daniel-bluesea, 2016-12-22:
what could be the cause for device->udid to be 40 characters long "ffffffffffffffffffffffffffffffffffffffff"?

nikias, 2016-12-22:
Can happen in Restore mode. Have seen this happening on incomplete previous restores, but also sometimes on first time restores. The open source usbmuxd is automatically adding a unique id (usb id), overwriting the last 4 bytes of the fffff* so if two devices are in the same state they will be distinguishable.

daniel-bluesea, 2016-12-25:
@nikias thanks, but..
on windows we dont use usbmuxd, but the libusbmuxd does what you describes also.
the problem is the for some reason sometime windows and/or its pnp driver or some usbhub hardware driver generates same device instance path (e.g. USB\VID_05AC&PID_12A8\FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF), and this is causing the BSOD..

daniel-bluesea, 2016-12-27:
it seems both windows7 & iPhone >= 6s "problem".
the iPhone devices when transition from recovery to restore gets this  "ffffffffffffffffffffffffffffffffffffffff" udid, and windows7 fails to generate unique device instance path for them.

any clues/thoughts how to overcome this situation?
this bsof is a problem when trying to restore multi devices..

mexmer, 2017-01-02:
USB\VID_05AC&PID_12A8\FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF as instance ID can only happen with bugged device firmware or driver.
Windows issue query for serial number to USB device (which may or may not be implemented - optional for everything except mass storage). If device responds with unknown/unimplemented command, windows generate "random" identification, but if device responds with value, windows use that value.

i've seen devices (although windows phones, and some chinesse flash drives - mostly fake capacity ones) that responded with weird numbers. although in case of apple udid, i would see it rather as issue with driver, especially if it's exactly 40F - never seen that personaly so far. but without communication capture, one can only guess.

nikias, 2017-01-02:
There's a simple solution: don't use Windows. Apple makes iOS and macOS. Yes, they made iTunes for Windows, but that's not their own platform, so it will never be perfect. They just made it work because so many people use Windows. But except for making it work with a single device, trust me they don't really care about Windows.
The only way to get it fixed is by reporting the issue with the driver to Apple and hope for them to fix it.

nikias, 2017-01-02:
Closing since the bug clearly doesn't originate from libusbmuxd.

closed 2017-01-02 by nikias

daniel-bluesea, 2017-01-02:
correct, it is a windows issue, there is a problem with some usb (3) drivers that fails to allocate/generate unique device instance path/id.
no problem here with the libusbmuxd.

thanks