libimobiledevice/libusbmuxd #45: I ran sudo make install but pkg-config can't find it

urherenow, 2017-01-03:
OS X Sierra. libelist seems to have built and installed fine, but this is preventing me from building libimobiledevice

Making install in common
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
Making install in src
 /opt/local/bin/gmkdir -p '/usr/local/lib'
 /bin/sh ../libtool   --mode=install /opt/local/bin/ginstall -c   libusbmuxd.la '/usr/local/lib'
libtool: install: /opt/local/bin/ginstall -c .libs/libusbmuxd.4.dylib /usr/local/lib/libusbmuxd.4.dylib
libtool: install: (cd /usr/local/lib && { ln -s -f libusbmuxd.4.dylib libusbmuxd.dylib || { rm -f libusbmuxd.dylib && ln -s libusbmuxd.4.dylib libusbmuxd.dylib; }; })
libtool: install: /opt/local/bin/ginstall -c .libs/libusbmuxd.lai /usr/local/lib/libusbmuxd.la
libtool: install: /opt/local/bin/ginstall -c .libs/libusbmuxd.a /usr/local/lib/libusbmuxd.a
libtool: install: chmod 644 /usr/local/lib/libusbmuxd.a
libtool: install: ranlib /usr/local/lib/libusbmuxd.a
make[2]: Nothing to be done for `install-data-am'.
Making install in include
make[2]: Nothing to be done for `install-exec-am'.
 /opt/local/bin/gmkdir -p '/usr/local/include'
 /opt/local/bin/ginstall -c -m 644  usbmuxd.h usbmuxd-proto.h '/usr/local/include/.'
Making install in tools
 /opt/local/bin/gmkdir -p '/usr/local/bin'
  /bin/sh ../libtool   --mode=install /opt/local/bin/ginstall -c iproxy '/usr/local/bin'
libtool: install: /opt/local/bin/ginstall -c .libs/iproxy /usr/local/bin/iproxy
make[2]: Nothing to be done for `install-data-am'.
make[2]: Nothing to be done for `install-exec-am'.
 /opt/local/bin/gmkdir -p '/usr/local/lib/pkgconfig'
 /opt/local/bin/ginstall -c -m 644 libusbmuxd.pc '/usr/local/lib/pkgconfig'
Michaels-iMac:libusbmuxd mike$ pkg-config --validate libusbmuxd
Package libusbmuxd was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusbmuxd.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusbmuxd' found

urherenow, 2017-01-03:
Ok, I manually fixed it.

first I tried: export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig and this worked, but only temporarily.

Permanent fix was to add the above line to /Users/<my name>/.profile right under the export path by MacPorts.

Why am I having to do this manually? Does a script need to be fixed? Do locations used for "make install" have to be adjusted for OS X Sierra?

I'll leave this opened until I have a reply, so the issue is known.

nikias, 2017-01-03:
@urherenow it depends on what environment you are using. In case you're using macports, the default pkgconfig location is probably /opt/local/lib/pkgconfig so it will be using that.
You can just run ```PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh```
I don't see any problem with that since you just have to do this once...

closed 2017-01-03 by nikias