libimobiledevice/libusbmuxd #46: Issue a release

ennui93, 2017-02-12:
Please issue a release to encapsulate the changes since 1.0.10.

chalin, 2018-12-09:
In https://github.com/libimobiledevice/libimobiledevice/commit/f1ee0fd8b2b1dd65c64f5a67e208efc708295caa#diff-67e997bcfdac55191033d57a16d1408aR21, libusbmuxd version 1.1.0 was specified as a requirement for libimobiledevice:

> LIBUSBMUXD_VERSION=1.1.0

@nikias, when will **release 1.1.0** be made available via https://github.com/libimobiledevice/libusbmuxd/releases?

Related:
- #60 
- https://github.com/flutter/flutter/issues/22595
- https://github.com/flutter/website/issues/1776

cc @kwalrath

tvolkert, 2018-12-14:
ping?

odebroqueville, 2018-12-22:
Sorry if this is a stupid question, but is anyone working on release 1.1.0 of libusbmuxd ? What are the changes required from 1.0.10 to 1.1.0 ?

nikias, 2018-12-22:
I am working on it. I know it has been ages since the last release but there are some small things I want to take care of first. I will see if I can finish it in the next weeks.

chalin, 2018-12-22:
Great! Could some of those small things be accomplished with community help? cc @sfshaza2

timsneath, 2019-03-15:
Just a gentle ping here, @nikias -- let us know if there's anything we can do to help. We have taken a dependency on this component for Flutter, and we have a slightly painful installation process because we have to tell people to install from HEAD. Lots of folk would be greatly in your debt if you could issue an update, even if it's just an interim step while you take care of the broader changes you want to make. 

Fingers crossed!

xgqfrms, 2019-03-18:
[image]

> what's wrong with this and how install the conflict libs?

LinusU, 2019-07-31:
@nikias any updates on this? 😬 Would be nice to not have to track HEAD commits...

nkavian, 2019-08-29:
Yet again, any updates?

Keruspe, 2019-09-16:
A 1.1.0 would really be nice indeed