libimobiledevice/libusbmuxd #49: No package 'libplist' found

Vinz87, 2017-05-21:
I previously installed libplist getting the following summary:
Configuration for libplist 2.0.0:
-------------------------------------------

  Install prefix ..........: /usr/local
  Debug code ..............: no
  Python bindings .........: no

  Now type 'make' to build libplist 2.0.0,
  and then 'make install' for installation.

I then issued the make and make install command, getting no errors.

After that, I ran autogen.sh of libusbmuxd, and I get the "No package 'libplist' found" error.

nikias, 2017-05-21:
Have you tried, as suggested by the configure output, to set ```PKG_CONFIG_PATH``` accordingly?
```
PKG_CONFIG_PATH=/usr/local/lib/pkgconfig ./autogen.sh
```

Vinz87, 2017-05-21:
Well I didn't know the command to do that, tried the one you wrote and it worked, thanks.
I suggest you to put it in the readme ;)

closed 2017-05-21 by Vinz87

TravelTrader, 2020-01-02:
I have the same problem at the moment:

```
configure: error: Package requirements (libplist++ >= 1.11) were not met:
No package 'libplist++' found
```

But the version `sudo apt install libplist-dev=2.0.0-2ubuntu1` is installed already.

The command `dpkg -L libplist-dev` leads to
```
/.
/usr
/usr/include
/usr/include/plist
/usr/include/plist/plist.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/libplist.pc
/usr/share
/usr/share/doc
/usr/share/doc/libplist-dev
/usr/share/doc/libplist-dev/copyright
/usr/lib/x86_64-linux-gnu/libplist.so
/usr/share/doc/libplist-dev/changelog.Debian.gz
```

So I tried `PKG_CONFIG_PATH=/usr/lib/pkgconfig ./autogen.sh`, `PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig ./autogen.sh` and `PKG_CONFIG_PATH=/usr/lib/x86_64/pkgconfig ./autogen.sh` but without any success.
I always get the above message at the end.

Thanks for any help!

TravelTrader, 2020-01-02:
Already solved.
I did only install `libplist-dev` without knowing that there's also a `libplist++-dev`.