libimobiledevice/libusbmuxd #54: Add option to bind iproxy socket to 0.0.0.0

NickAb, 2018-06-21:
Executing `iproxy LOCAL_TCP_PORT DEVICE_TCP_PORT` will bind LOCAL_TCP_PORT to localhost only.

Make an optional flag to bind to 0.0.0.0 so that the port will be exposed and accessible by other hosts.
This is useful to build small device-farm, where all devices connected to one host and other hosts are used to run tests, etc. on devices connected remotely.

Current workaround is to use `socat`

```bash
socat tcp-listen:LOCAL_TCP_PORT,reuseaddr,fork tcp:0.0.0.0:LOCAL_TCP_PORT
```

But this requires extra dependencies to be installed on host and managing extra process per port.

Ad1991, 2019-08-16:
A PR to fix this issue #74 which enables accepting connections from other machines in network.

nikias, 2023-04-22:
Support for this was added with commit https://github.com/libimobiledevice/libusbmuxd/commit/910166b3d67652abb54dece7e1cacdc9dcfdb659.

closed 2023-04-22 by nikias