libimobiledevice/libusbmuxd #56: idevice_id -l command lists same udid twice

aryalakshmi, 2018-10-08:
Hi,

I am facing the issue that ideviceinstaller shows same udid twice. 
The 2 USB ports in MAC lists one UDID each.
If i unplug anyone of the USB , one out the list disappears. 
Backup sync in the IphoneX is switched off.

Below is my setup.
MAC ---USB --- IphoneX(Connected to Router Wifi)
Also, MAC---USBport--- EthernetPort of Router

Can someone please help me with this issue.

Thanks,
Arya

aryalakshmi, 2018-10-08:
i tried all of below solutions
1)Disable sync over wifi
2) delete .plist for the udid from /var/db/lockdown/ 
I am not able to try the solution since i am not aware of which interface to use. https://github.com/libimobiledevice/libimobiledevice/issues/152#issuecomment-420402149

Please help!

Thanks,
Arya

harlentan, 2019-02-01:
> i tried all of below solutions
> 1)Disable sync over wifi
> 2) delete .plist for the udid from /var/db/lockdown/
> 3) [libimobiledevice/libimobiledevice#152 (comment)](https://github.com/libimobiledevice/libimobiledevice/issues/152#issuecomment-420402149)
> 
> Please help!
> 
> Thanks,
> Arya

Any solutions updated? I also face the same issue.

nikias, 2020-05-26:
This is fixed by now.

closed 2020-05-26 by nikias