libimobiledevice/libusbmuxd #61: Getting error when trying to compile

plato79, 2018-12-27:
Here is what I get:
```
...
Making all in tools
  CCLD     iproxy
ld: warning: ignoring file ../common/.libs/libinternalcommon.a, file was built for archive which is not the architecture being linked (x86_64): ../common/.libs/libinternalcommon.a
Undefined symbols for architecture x86_64:
  "_socket_accept", referenced from:
      _main in iproxy-iproxy.o
  "_socket_close", referenced from:
      _main in iproxy-iproxy.o
      _acceptor_thread in iproxy-iproxy.o
      _run_ctos_loop in iproxy-iproxy.o
      _run_stoc_loop in iproxy-iproxy.o
  "_socket_create", referenced from:
      _main in iproxy-iproxy.o
  "_socket_receive_timeout", referenced from:
      _run_ctos_loop in iproxy-iproxy.o
      _run_stoc_loop in iproxy-iproxy.o
  "_socket_send", referenced from:
      _run_ctos_loop in iproxy-iproxy.o
      _run_stoc_loop in iproxy-iproxy.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

What could be the problem?!? I'm using OSX 10.13.6.
Xcode 10.1 and Homebrew are installed in the system..

ChiChou, 2019-02-12:
#440

ChiChou, 2019-02-12:
https://github.com/libimobiledevice/libimobiledevice/issues/440

nikias, 2019-05-26:
This is the problem: 
```
ld: warning: ignoring file ../common/.libs/libinternalcommon.a, file was built for archive which is not the architecture being linked (x86_64): ../common/.libs/libinternalcommon.a
```
A `make clean` followed by `make` should fix the issue.

closed 2019-05-26 by nikias

eran-, 2019-09-10:
I'm having the same issue now on a9e69b1 as well. `make clean && make` didn't change the behavior.