libimobiledevice/libusbmuxd #66: segfault when attaching iPhone

erjiang, 2019-05-24:
```
[ 8390.186788] usb 3-6: new high-speed USB device number 10 using xhci_hcd
[ 8390.336132] usb 3-6: New USB device found, idVendor=05ac, idProduct=12a8
[ 8390.336137] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8390.336140] usb 3-6: Product: iPhone
[ 8390.336142] usb 3-6: Manufacturer: Apple Inc.
[ 8390.336145] usb 3-6: SerialNumber: 9a2*elided*
[ 8390.386199] ipheth 3-6:4.2: Apple iPhone USB Ethernet device attached
[ 8390.394531] ipheth 3-6:4.2 enp0s20u6c4i2: renamed from eth0
[ 8390.395001] usbmuxd[23403]: segfault at fffffffffffffff9 ip 00007f0c61bc198d sp 00007f0c5f91bdb0 error 5 in libc-2.27.so[7f0c61b2a000+1e7000]
[ 8390.421028] IPv6: ADDRCONF(NETDEV_UP): enp0s20u6c4i2: link is not ready
[ 8390.421523] IPv6: ADDRCONF(NETDEV_UP): enp0s20u6c4i2: link is not ready
```

Using latest git revision 8b82ef1 compiled on Ubuntu 18.04. Please let me know what next steps I should take to debug.

nikias, 2019-05-25:
The segfault is in usbmuxd. Can you check if you are also using the latest code for **usbmuxd** https://github.com/libimobiledevice/usbmuxd ? If no, please update and test again, if yes or if it still crashes, can you run it from gdb?
```
sudo killall usbmuxd
```
```
sudo gdb usbmuxd -f
```
Then plug the device and see if you still get a segfault. If yes, type `bt` to get a backtrace, and paste it here.

erjiang, 2019-06-04:
```
(gdb) run -f
Starting program: /usr/sbin/usbmuxd -f
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[16:25:06.860][3] usbmuxd v1.1.0 starting up
[New Thread 0x7ffff61ef700 (LWP 20558)]
[16:25:06.864][3] Initialization complete
[16:25:15.873][3] Could not get old configuration descriptor for device 3-10: -5
[16:25:15.887][3] Connecting to new device on location 0x3000a as ID 1
[16:25:15.888][3] Connected to v2.0 device 1 on location 0x3000a with serial number 9a2df180bdeffc2243e17d01bdd7fc7fe81bf95b
[New Thread 0x7ffff59ee700 (LWP 20610)]
[16:25:15.895][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device 9a2df180bdeffc2243e17d01bdd7fc7fe81bf95b, lockdown error -8

Thread 3 "usbmuxd" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff59ee700 (LWP 20610)]
__GI___libc_free (mem=0x1) at malloc.c:3103
3103	malloc.c: No such file or directory.
(gdb) bt
#0  __GI___libc_free (mem=0x1) at malloc.c:3103
#1  0x00007ffff79a19df in idevice_free () from /usr/local/lib/libimobiledevice.so.6
#2  0x000055555555c445 in ?? ()
#3  0x00007ffff75736db in start_thread (arg=0x7ffff59ee700) at pthread_create.c:463
#4  0x00007ffff729c88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

I realize I forgot to post info about the iPhone:

Software Version: 12.2
Model Name: iPhone 7
Model Number: MN8M2LL/A

holgerengels, 2019-09-26:
seeing exactly the same segfault ..

`
(gdb) run -f
Starting program: /usr/sbin/usbmuxd -f
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[21:18:18.989][3] usbmuxd v1.1.1 starting up
[New Thread 0x7ffff75b6700 (LWP 9536)]
[21:18:18.991][3] Initialization complete
[21:18:18.992][3] Connecting to new device on location 0x30005 as ID 1
[21:18:18.992][3] Connected to v2.0 device 1 on location 0x30005 with serial number 6bb8e253f9849961d336d0ad31d827bbefb2a946
[New Thread 0x7ffff6db5700 (LWP 9538)]
[21:18:18.992][1] preflight_worker_handle_device_add: ERROR: Could not connect to lockdownd on device 6bb8e253f9849961d336d0ad31d827bbefb2a946, lockdown error -8

Thread 3 "usbmuxd" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff6db5700 (LWP 9538)]
__GI___libc_free (mem=0x1) at malloc.c:3085
3085    malloc.c: No such file or directory.
(gdb) bt
#0  __GI___libc_free (mem=0x1) at malloc.c:3085
#1  0x00007ffff7d743cf in idevice_free (device=0x7ffff0000b20) at idevice.c:301
#2  0x000055555555c7e5 in ?? ()
#3  0x00007ffff7d42164 in start_thread (arg=<optimized out>) at pthread_create.c:486
#4  0x00007ffff7c6adef in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
`
mine is an iPad Air 2 running 12.4.1

what can I do about it?

nikias, 2023-04-22:
I think this was recently fixed in usbmuxd. Feel free to reopen an issue on usbmuxd if the issue persists.

closed 2023-04-22 by nikias