libimobiledevice/libusbmuxd #67: Remove unnecessary null pointer checks

elfring, 2019-05-26:
[An extra null pointer check is not needed](https://pubs.opengroup.org/onlinepubs/9699919799/functions/free.html "Do I need to check for null before free p?") in functions like the following.
- [get_next_event](https://github.com/libimobiledevice/libusbmuxd/blob/7430eec6dc7251275559dd4705aaa068af49e5f6/src/libusbmuxd.c#L1077)
- [usbmuxd_get_result](https://github.com/libimobiledevice/libusbmuxd/blob/7430eec6dc7251275559dd4705aaa068af49e5f6/src/libusbmuxd.c#L501)

nikias, 2019-05-26:
Yes I will fix that too.

nikias, 2019-05-26:
Fixed with 5bd193ff94f9070cfd9e64c5502574d13750a152.

closed 2019-05-26 by nikias

elfring, 2019-05-26:
Thanks for [your small source code improvement](https://github.com/libimobiledevice/libusbmuxd/commit/5bd193ff94f9070cfd9e64c5502574d13750a152 "Removal of two unnecessary checks").
[Another update candidate was left over](https://github.com/libimobiledevice/libusbmuxd/blob/5bd193ff94f9070cfd9e64c5502574d13750a152/src/libusbmuxd.c#L1076 "Update candidate: get_next_event()").

nikias, 2019-05-26:
Fixed.

elfring, 2019-05-26:
Thanks for [another small source code improvement](https://github.com/libimobiledevice/libusbmuxd/commit/7b007832f6e6106f2fb2eba896efc69e38dfe374 "Removal of four unnecessary checks").