libimobiledevice/libusbmuxd #68: gcc-9.1 compile error

sanitariu, 2019-05-27:
make[2]: Entering directory '/root/iphone/libusbmuxd/src'
  CC       libusbmuxd.lo
libusbmuxd.c:865:12: error: static declaration of 'pselect' follows non-static declaration
  865 | static int pselect(int nfds, fd_set readfds, fd_set writefds, fd_set exceptfds, const struct timespec timeout, const sigset_t *sigmask)
      |            ^~~~~~~
In file included from /usr/include/sys/types.h:196,
                 from /usr/include/stdlib.h:394,
                 from libusbmuxd.c:27:
/usr/include/sys/select.h:113:12: note: previous declaration of 'pselect' was here
  113 | extern int pselect (int _nfds, fd_set *__restrict _readfds,
      |            ^~~~~~~

I changed the following:

static int pselect(int nfds, fd_set *readfds, fd

to

int pselect(int nfds, fd_set *readfds, fd

Now libusbmuxd compiles fine.

nikias, 2019-05-27:
That's not a correct fix. Problem here is that configure didn't detect pselect availability. Can you run configure again and see if it detects pselect? If so config.h will have a #define HAVE_PSELECT 1  otherwise please could you post config.log somewhere (like pastebin)? Thanks.

cshlxm, 2019-05-27:
I have same error ,and pselect  has not been detected  ,and there was no  #define HAVE_PSELECT 1   in config.h

sanitariu, 2019-05-27:
Here is the logfile:
https://pastebin.com/hyD48nQt

I think here is why it uses system pselect:
#ifdef HAVE_INOTIFY
#include <sys/inotify.h>
#include <sys/select.h>

And in select.h we have it already:
extern int pselect (int __nfds, fd_set *__restrict __readfds,
                    fd_set *__restrict __writefds,
                    fd_set *__restrict __exceptfds,
                    const struct timespec *__restrict __timeout,
                    const __sigset_t *__restrict __sigmask);

Hiepars, 2019-05-27:
i have same issue.

sanitariu, 2019-05-27:
ok which is the right solution :) ?

nikias, 2019-05-28:
The right solution is to figure out why it doesn't detect pselect as being available

nikias, 2019-05-28:
Uh my bad, looks like I forgot to commit the change in configure.ac - fixed with commit 60109fdef47dfe0badfb558a6a2105e8fb23660a.

closed 2019-05-28 by nikias