libimobiledevice/libusbmuxd #71: Commit ca24570 breaks `flutter run` command

christopherfujino, 2019-06-26:
There are [multiple] [reports] [of] the latest HEAD (since [ca24570](https://github.com/libimobiledevice/libusbmuxd/commit/ca24570992df0bd5a0c6be26439acbb1e0523c5f)) breaking flutter run. Could you please revert?

JonGabilondoAngulo, 2019-06-27:
I can confirm that this push is braking our communication to the device too.
Reverting the push solves the problem.

nikias, 2019-06-28:
@christopherfujino I don't know how flutter is using the library exactly. It's a very simple fix and instead of demanding a revert of a mandatory change it would only be straight forward to fix the issue on the other side, in this case flutter.
I realized though that iproxy was broken in fact, which I just fixed with commit 6539b0261889bff41972338fa64cd230183fad42. If flutter is only using iproxy, then this would solve the problem.
@JonGabilondoAngulo you don't even mention how you do your communication.
As a general rule to adapt to the changes:
* Before commit ca24570, a timeout was basically a `usbmuxd_recv` or `usbmuxd_recv_timeout` with a return value of 0 (success) while setting the number of bytes received (via recv_bytes parameter) to 0.
* Starting with commit ca24570, code expecting a timeout needs to check for an error code of `-ETIMEDOUT` to know if a timeout occurred or not. But also recv_bytes will be 0 in that case.

amake, 2019-06-30:
I can't speak as to exactly how Flutter uses libusbmuxd, but 6539b02 does indeed appear to fix the issue. Thank you!

christopherfujino, 2019-07-01:
Thanks @nikias, closing this issue as it appears resolved!

closed 2019-07-01 by christopherfujino