libimobiledevice/libusbmuxd #78: Buffers limit transfer speed

cbrx-fw, 2019-09-16:
Increase the size of the send and receive buffers to improve performance of the network connections. Please see the attached patch. This was found when testing with Cambrionix ThunderSync2-16 and idevicerestore and achieved a speedup that halved the time taken to restore 16 iPads.
[idev_sock_bufsize.txt](https://github.com/libimobiledevice/libusbmuxd/files/3615907/idev_sock_bufsize.txt)

nikias, 2019-09-16:
Nice. I already have this in git: https://github.com/libimobiledevice/libusbmuxd/blob/master/common/socket.c#L144-L150 maybe you are looking at an older source version?

cbrx-fw, 2019-09-16:
Nice. I already have this in git: https://github.com/libimobiledevice/libusbmuxd/blob/master/common/socket.c#L144-L150 maybe you are looking at an older source version?

Yes, so you do, sorry for the noise. That’s great.

Thanks,

Andrew

nikias, 2019-09-20:
No worries! Thanks for the feedback :)

closed 2019-09-20 by nikias