libimobiledevice/libusbmuxd #81: soname tracking

corsac-s, 2019-10-05:
Hi,

it seems that a recent commit changed libusbmuxd ABI: https://github.com/libimobiledevice/libusbmuxd/commit/f5a7387a54ae08c9cd1d83a415393e0e909dc6e6#diff-038b1c435b1baca0fc7faaf5e375f401L37 added a field in `usbmuxd_device_info_t`, which means applications and libraries built against an old version of libusbmuxd will crash when running with an updated version.

This as caused bugs in Debian (https://bugs.debian.org/941703) where upower and tools in libimobiledevice are crashing.

Would it be possible to do dependency tracking in exposed symbols, and update the soname/soversion when that happens?

nikias, 2019-10-05:
Yes, totally. I also added at least one function to the interface, so for that the SO_VERSION should be changed 5:0:1 => 6:0:2 plus the internal changes => 6:1:2, does that make sense?

corsac-s, 2019-10-06:
I'm not an expert in symbol versioning. I /think/ since the struct change and the added function appeared in the same commit, it should be ok to go 6:0:2 but don't quote me on that. For the added function we don't have issues in Debian because we use symbol versioning (so we know that something built against a version with the new function will need that same version or later to run).

corsac-s, 2019-10-06:
And this is likely the cause for https://github.com/libimobiledevice/ifuse/issues/57

corsac-s, 2019-10-22:
Any news on this?

corsac-s, 2019-10-22:
And actually I think that when you bump `current` you're supposed to reset `revision` and `age` to 0.

nikias, 2019-10-22:
Yeah I don't know, maybe @cfergeau has an idea?

nikias, 2019-10-22:
Thing is, I added stuff to the interface but kept the others. I didn't remove anything, and the semantic didn't really change I think... so it's +1 : 0 : +1  from what I understand but yeah... I dunno 100% and I want to do it right.

corsac-s, 2019-10-22:
https://www.gnu.org/software/libtool/manual/html_node/Updating-version-info.html indicates:

> Start with version information of ‘0:0:0’ for each libtool library.
> Update the version information only immediately before a public release of your software. More frequent updates are unnecessary, and only guarantee that the current interface number gets larger faster.
> If the library source code has changed at all since the last update, then increment revision (‘c:r:a’ becomes ‘c:r+1:a’).
> **If any interfaces have been added, removed, or changed since the last update, increment current, and set revision to 0.**
> If any interfaces have been added since the last public release, then increment age.
> If any interfaces have been removed or changed since the last public release, then set age to 0. 

So I'd say current+=1 and revision=0. With the last two rules I'm unsure about age.

corsac-s, 2019-10-22:
A possibility would be to just bump current when the ABI changes in incompatible ways and keep age=current=0. Ideally that would just be done pre-release, but if there are no releases I guess it should be done when an ABI change is committed.

nikias, 2019-10-22:
Well I just have to make a new release and then when such changes incur do a new release. Just need to know what's the right soversion now :)

corsac-s, 2019-10-24:
Then I'd say just go ahead with 6:0:0 (and bump to 7:0:0 next time ABI changes incompatibly, and so-on).

corsac-s, 2019-10-31:
Any news on this? I had a _go_ from the Debian release team to proceed with the soname change in order to unbreak current installs, but I don't want to divert from upstream.

corsac-s, 2019-11-15:
Hey, sorry to bother you again but this issue is still biting people in Debian unstable and testing

nikias, 2019-11-16:
See tag 2.0.0

corsac-s, 2019-11-16:
Thanks, will take a look!

nikias, 2019-11-26:
I think this should be fine now.

closed 2019-11-26 by nikias