libimobiledevice/libusbmuxd #84: icat: collides with sleuthkit

blshkv, 2019-11-24:
both libusbmuxd and sleuthkit install icat binary into /usr/bin. That makes it harder to main the tool.
It also might make break other tools which call "/usr/bin/icat".

Please change the name to something more unique.

nikias, 2019-11-24:
I can't really think about a better name especially now that it's in use already...
@aguinet what do you think about 'muxcat'?

aguinet, 2019-11-24:
I'm fine with both `inetcat` or `muxcat`. `idevcat` could also be a solution!

nikias, 2019-11-24:
Oh I didn't see the inetcat suggestion. I think that's probably the name to go with. Let me make a poll on twitter.

aguinet, 2019-11-25:
BTW, the suggested patch does not seem correct to me. It supposes libusbmuxd is installed in `/usr` (which might not always be the case), and would erase a potentially already existing `icat` binary.

The proper patch IMHO is just to rename the name of the tool within the source code!

nikias, 2019-11-26:
Done: 2b452316ccdab081cd6add860d6f0eaab46c4879 tag 2.0.1

closed 2019-11-26 by nikias

nikias, 2019-11-26:
Sorry, made a mistake, new commit is e97cebeaf7d8f3793b40c3e7bb3ab63853098f5d.