libimobiledevice/libusbmuxd #88: Cannot find iPhone via network (idevice_id -n)

fosple, 2020-02-09:
I saw in https://github.com/libimobiledevice/libusbmuxd/blob/master/NEWS that network support was added to libusbmuxd and I'm super happy about that fact 👍 

However, I don't get it working. I've compiled and installed the latest version of all the needed libraries (libimobiledevice, libplist, libusbmuxd, usbmuxd) from source on my Raspberry Pi 4 (OS: Debian Buster).

Via USB the IPhone shows up. So I connected the iPhone via USB and paired it (idevicepair pair) before I tried to connect via network.

However, after unplugging the iPhone from USB and trying to connect to it via the network I didn't get any result:

```
sudo usbmuxd -s -v
[19:56:08.141][3] usbmuxd v1.1.1 starting up
[19:56:08.141][4] Creating socket
[19:56:08.142][4] Initializing USB
[19:56:08.159][4] Registering for libusb hotplug events
[19:56:08.159][4] 0 devices detected
[19:56:08.159][3] Initialization complete
[19:56:08.159][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.                                                      
[19:56:11.902][4] New client on fd 11 (pid 16768)
[19:56:11.904][4] Client 11 connection closed
[19:56:11.904][4] Disconnecting client fd 11
```

```
idevice_id -n
# no result
```

Further infos:
- avahi-daemon is started and running
- ufw firewall is disabled (I thought maybe I blocked some ports accidentally)

Would be super nice if you can help me :)

nikias, 2020-05-26:
I am working on network/mDNS capabilities for the open source usbmuxd.

alexandre-janniaux, 2020-08-20:
> I am working on network/mDNS capabilities for the open source usbmuxd. This will still take some time to get it done properly.

Hi, did you start your work on this? May I provide help and patches for it?

reanimus, 2020-11-18:
Any update here? I'd love to get Wifi Sync working on Linux :)

cemizm, 2020-12-13:
> Any update here? I'd love to get Wifi Sync working on Linux :)

+1

cemizm, 2020-12-24:
hi @nikias, any progress on network/mDNS capabilities? would love to support you on this topic, but don't know where to start... are there any developer documentations or at least a high level architecture overiew of libimobiledevice?

mcdull, 2021-01-15:
May I confirm with wi-fi sync only works in Mac but not Linux?

fosple, 2021-05-11:
@mcdull WiFi sync works on Mac and Win, but not yet on Linux as there is no open source version of usbmuxd yet which supports WiFi sync.

voltagex, 2021-05-15:
@fosple can you confirm this is still the case with the Git master version of the code?

bentumbler, 2021-05-26:
I have managed to get it working on a rPi although a little flaky sometimes.
I did install https://github.com/tihmstar/usbmuxd2 though. Not sure if I could have got it working without that.

fosple, 2021-05-26:
Wow, thanks @bentumbler for pointing out the re-implementation of usbmuxd by @tihmstar https://github.com/tihmstar/usbmuxd2 . Maybe @nikias can use it and include it in the original project 🥳 

I'll give usbmuxd2 a try on the weekend.

kozfogel, 2021-12-26:
Hello guys,

@bentumbler - i tried to build usbmuxd2 on a rPi, but somehow i cannot make it work.
I followed this tutorial:
https://www.numerousnetworks.co.uk/uncategorized/untether-ios-step-by-step/

but when at the point of making usbmuxd2 i got following problem

Client.cpp:274:50: error: ‘plist_to_bin_free’ was not declared in this scope; did you mean ‘plist_to_bin’?
  274 |                         safeFreeCustom(plistbin, plist_to_bin_free);
      |                                                  ^~~~~~~~~~~~~~~~~
      
Do you have some information how you got it to work? Or any other advice

Thanks

bentumbler, 2021-12-26:
It's possible some of the dependencies have moved on a version. Did you install them all, especially both the libplist dev dependencies

https://www.numerousnetworks.co.uk/uncategorized/untether-ios-step-by-step/

kozfogel, 2021-12-26:
i installed libplist-dev and libplist++-dev (via apt-get) and builded libplist from git (because version 2.2.1 is needed)

cemizm, 2021-12-27:
Hi kozfogel, 

installing the apt packages are unnecessary if you build libplist from git. the install step of the build process will install these.

had the same issue with the plist_to_bin_free function. it has been dropped a few days ago: 
https://github.com/libimobiledevice/libplist/commit/c31beaaec196c26990139e3fc4f89f996d7b86e9

In order to solve this issue you should checkout commit a9e34bd29ae9dcdae55bdf5fb8a23c9b1c02eee9

juliienp, 2021-12-31:
Hi, i'm trying to build a docker image to run libimobiledevice and sync via Wifi every night.

I got some issues when I build usbmuxd2..

I also tried to follow these guide : https://www.numerousnetworks.co.uk/uncategorized/untether-ios-step-by-step/

Here's my dockerfile : 
`
FROM ruby:2.3-slim

RUN apt-get update \
  && apt-get install -y \
    build-essential \
    checkinstall \
    git \
    autoconf \
    automake \
    libtool-bin \ 
    python-dev

RUN apt-get install -y \
  libplist-dev \
  libusbmuxd-dev \
  libimobiledevice-dev \
  libusb-1.0-0-dev \
  libplist++-dev \
  libssl-dev \
  usbmuxd \
  udev \
  libavahi-client-dev \
  avahi-utils \
  libatomic-ops-dev \
  libatomic1 \
  bluez-tools

ENV INSTALL_PATH /src
RUN mkdir -p $INSTALL_PATH

WORKDIR $INSTALL_PATH
RUN git clone https://github.com/libimobiledevice/libplist.git \
  && cd /src/libplist \
  && git checkout a9e34bd29ae9dcdae55bdf5fb8a23c9b1c02eee9 \
  && ./autogen.sh \
  && make \
  && make install

RUN ldconfig

WORKDIR $INSTALL_PATH
RUN git clone https://github.com/tihmstar/libgeneral.git \
  && cd libgeneral \
  # && git checkout 55  \
  && ./autogen.sh \
  # && make CFLAGS="-g -O2 -std=c11 -latomic" LDFLAGS=-latomic \
  && make install

RUN ldconfig

WORKDIR $INSTALL_PATH
RUN git clone https://github.com/tihmstar/usbmuxd2.git  \
  && cd /src/usbmuxd2 \
  && git submodule init \
  && git submodule update \
  && ./autogen.sh \
  && make \
  && make install
  
RUN ldconfig

WORKDIR $INSTALL_PATH
RUN git clone https://github.com/libimobiledevice/libimobiledevice.git \
  && cd /src/libimobiledevice \
  # && git checkout 1.3.0 \
  && ./autogen.sh \
  && make \
  && make install

RUN apt-get -y install ideviceinstaller
`

Do you know where I misunderstood something ?

voltagex, 2021-12-31:
> Hi, i'm trying to build a docker image to run libimobiledevice and sync via Wifi every night.
>
> I got some issues when I build usbmuxd2..
>
> I also tried to follow these guide : [...]

1. You have replied on an unrelated issue
2. You have not posted an error message or problem

bentumbler, 2022-01-02:
I haven't tried to replicate your issue but I do see that you have not built libusbmuxd as per the instructions at https://www.numerousnetworks.co.uk/uncategorized/untether-ios-step-by-step/

Also, I found that to build usbmuxd2 with g++ 8.3 I needed to amend configure.ac with the following `LDFLAGS+="-latomic -lstdc++fs"`

Try those suggestions. Also follow the build order in the guide.
If you still have issues, we need to see the build output.

cemizm, 2022-01-02:
Here is my dockerfile which seems to work after a lot of fiddling and version matching..

https://gist.github.com/cemizm/07ed041d451c55e395856873f46f074f

It uses a small bash script as an entrypoint in order to the required services. furthermore for the first usage you need to (re-)start usbmuxd a few times in order to create the SystemConfiguration.plist (see issue https://github.com/tihmstar/usbmuxd2/issues/6)

for the network usage of a device you need to connect the device at least once trough usb, which means you need to pass the usb bus to the container and run it in privileged mode. the public and private keys after the pairing process are stored in the data folder and are persisted between startups.

**Usage:**

 ```bash
#build the container
docker build -t usbmuxd2

#run in default mode
docker run -it --rm --name imobiledevice -v $(PWD)/data:/var/lib/lockdown/ --network host usbmuxd2

#run with usb support and privileged mode
docker run -it --rm --name imobiledevice -v $(PWD)/data:/var/lib/lockdown/ --network host --privileged -v /dev/bus/usb:/dev/bus/usb usbmuxd2
```

kozfogel, 2022-01-02:
@bentumbler @cemizm 

Thanks a lot guys. I got it running on my PI4 now (i used the Docker script with some slightly changes for latomic)

It works when i connect via USB - but Wifi sadly not. 
Any idea where to look next? (logs, debuging, etc)
usbmuxd output:

[16:28:02.685][3] TCP Connected to device
[16:28:02.685][1] failed to recv_data on client -1 with error=graceful kill code=25165834
[16:28:02.686][3] killing client -1
[16:28:02.690][1] failed to recv_data on client 17 with error=client 17 disconnected! code=6946826
[16:28:02.690][3] killing client 17
[16:28:02.705][1] failed to recv_data on client 18 with error=client 18 disconnected! code=6946826
[16:28:02.705][3] killing client 18
[16:28:02.921][3] preflight_device: Finished preflight on device 00008101-00090DAE2881401D
[16:28:02.923][3] killing TCP 16
[16:28:05.694][3] killing device 00008101-00090DAE2881401D
[16:28:05.994][1] failed to recv_data on client 17 with error=client 17 disconnected! code=6946826
[16:28:05.994][3] killing client 17
[16:28:10.998][1] Failed to start WIFIDevice 00008101-00090DAE2881401D with error=4128790 ([WIFIDevice] Failed to start heartbeat service with error=-256)
[16:28:10.999][3] killing device 00008101-00090DAE2881401D
[libusbmuxd] device_info_from_plist: Failed to get EscapedFullServiceName!
[16:28:11.004][1] Failed to start WIFIDevice 00008101-00090DAE2881401D with error=3997718 (assure failed)
[16:28:11.004][1] failed to recv_data on client 14 with error=client 14 disconnected! code=6946826
[16:28:11.004][3] killing device 00008101-00090DAE2881401D
[16:28:11.004][3] killing client 14
[16:29:27.704][1] failed to recv_data on client 15 with error=client 15 disconnected! code=6946826
[16:29:27.705][3] killing client 15
[16:29:27.705][3] Starting TCP connection
[16:29:27.706][3] TCP Connected to device
[16:29:27.707][1] failed to recv_data on client -1 with error=graceful kill code=25165834
[16:29:27.707][3] killing client -1
[16:29:27.711][1] failed to recv_data on client 14 with error=client 14 disconnected! code=6946826
[16:29:27.711][3] killing client 14
[16:29:27.712][1] failed to recv_data on client 15 with error=client 15 disconnected! code=6946826
[16:29:27.712][3] killing client 15
[16:29:27.719][1] failed to recv_data on client 16 with error=client 16 disconnected! code=6946826
[16:29:27.720][3] killing client 16
[16:29:27.967][3] killing TCP 10
[16:29:32.047][3] Device 1-5 RX aborted due to error or disconnect
[16:29:32.048][3] Device 1-5 RX aborted due to error or disconnect

cemizm, 2022-01-02:
> @bentumbler @cemizm
> 
> Thanks a lot guys. I got it running on my PI4 now (i used the Docker script with some slightly changes for latomic)
> 

it seems like you are not using the latest version (> 1.3.0) of libimobiledevice... please make sure that your are not using any packages (libimobiledevice, libplist, ...) from a package manager.

can you post your dockerfile?

kozfogel, 2022-01-02:
I am not running docker. I builded all directly on the Pi - Raspbian GNU/Linux 11 (bullseye) - fresh install.

if i am running somithing like this:
ideviceinfo --version
ideviceinfo 1.3.0

should it be a higher version?

bentumbler, 2022-01-02:
Did you follow the instructions related to avahi? Network connections require mDNS. Your network also needs to support mDNS. Most ione networks do

mexmer, 2022-01-03:
> I am not running docker. I builded all directly on the Pi - Raspbian GNU/Linux 11 (bullseye) - fresh install.
> 
> if i am running somithing like this: ideviceinfo --version ideviceinfo 1.3.0
> 
> should it be a higher version?

yes, it should be 1.3.1, version 1.3.0 is year and half old, and supports only devices up to ios 13

bentumbler, 2022-01-03:
V 1.3.0 works fine for my iOS 15 devices (at least for log capture, which is how I use it). You can move to a later version but I think you will need to ensure you build the libimobiledevice-glue library too.

mexmer, 2022-01-04:
@bentumbler version 1.3.0 works only for limited set of features, some services changed their required protocol version (plus other minor changes), and those are updated in 1.3.1, screenshot is one of them, also backup, sync ... not sure what others. guess i should have written **full support**

dmitrym0, 2022-09-06:
@kozfogel did you manage to resolve your issue? I'm having something similar:

```
[13:46:23.385][5] add_device 0000811000163D5A2684801E
[13:46:23.385][5] Muxer: adding device 0000811000163D5A2684801E assigning id 3
[13:46:23.385][5] [Client] initializing Client 16
[13:46:23.385][5] add_client 16
[13:46:23.385][5] Client command in fd 16 len 436 ver 1 msg 8 tag 2
[13:46:23.385][5] send_pkt fd 16 tag 2 msg 8 payload_length 946
[13:46:23.385][1] Failed to start WIFIDevice 0000811000163D5A2684801E with error=3801110 (assure failed)
[13:46:23.385][5] delete_device 0000811000163D5A2684801E
[13:46:23.385][5] delete_device 0000811000163D5A2684801E
[13:46:23.385][1] failed to recv_data on client 16 with error=client 16 disconnected! code=6750218
```

Looks like the device is recognized, but errors out anyway. At least it's the correct UDID.

@mexmer where can one obtain version 1.3.1?

```
ideviceinfo --version
ideviceinfo 1.3.0-164-gb5ce444
```

looks like the version string is set using `git describe`. I don't see a tag for 1.3.1.

mexmer, 2023-08-02:
@sepsky usbmuxd2 is reimplementation of usbmuxd with avahi support, so you need to direct any issue to author of usbmuxd2
https://github.com/tihmstar/usbmuxd2

mexmer, 2023-08-02:
on windows it works, so it's very likely something on usbmuxd2 side. although there have been some changes in source code, if you look at commits it's mostly cleanup, nothing that should cause communication break with usbmuxd (or usbmuxd2)

sepsky, 2023-08-08:
Thanks @mexmer for the encouraging comment. 

I think I have found the issue, but still not sure about the solution :( 

So I think it's something to do with dns and socket configuration (either on pi, ATV or the network ?) 
I saw @bentumbler  comment regarding the dns configuration, but given that my experience is apple-only (apple to apple devices, all "works" approach), I'm not well-versed in sockets and how to connect to apple devices with non-apple devices. 

I'm learning about that, but if you (or @bentumbler ) have any hints / help you think would be useful for that topic, I'd greatly appreciate it.

cl532257933, 2023-08-14:
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[20:56:13.063][3] starting usbmuxd2 version: 0.46-753b79eaf317c56df6c8b1fb6da5847cc54a0bb0
[20:56:13.063][3] Loaded config
[20:56:13.063][3] starting usbmuxd2 version: 0.46-753b79eaf317c56df6c8b1fb6da5847cc54a0bb0
[New Thread 0x7ffff70c1640 (LWP 4144)]
[20:56:13.064][3] Inited ClientManager
[20:56:13.064][3] USBDeviceManager libusb 1.0
[New Thread 0x7ffff68c0640 (LWP 4145)]
[New Thread 0x7ffff60bf640 (LWP 4146)]
[20:56:13.065][3] Registering for libusb hotplug events
[New Thread 0x7ffff58be640 (LWP 4147)]
[20:56:13.066][3] Inited USBDeviceManager
[New Thread 0x7ffff50bd640 (LWP 4148)]
[20:56:13.067][3] Inited WIFIDeviceManager
[20:56:13.068][4] Initialization complete
[New Thread 0x7ffff48bc640 (LWP 4150)]
[20:56:36.652][1] failed to recv_data on client 9 with error=client 9 disconnected! code=6750218
[Thread 0x7ffff48bc640 (LWP 4150) exited]

who can help me，i build it on ubuntu，but “”failed to recv_data on client 9 with error=client 9 disconnected! code=6750218

sepsky, 2023-08-15:
I have the same issue @cl532257933. 

I followed https://www.numerousnetworks.co.uk/guides/untether-ios-step-by-step/ (I had to make some changes to the code because there were some duplication). However, that's also giving me the same error. 

How did you install dependencies? Was it a mix of `apt-get` and manual compiling and installing (using `make` and `make install`...)? I really think this is something to do with the dependencies (I put my money on libplist or libplist++ because when I added a console output of what's passed to `recv_data`, we have some bytes. This points me to either an incorrect formatting of the data we receive, OR broken communication between `Avahi` and `usbmuxd`). 

I wonder if you also get the same error if you follow the above link? 

I'll update as soon as I found more.

bentumbler, 2023-08-15:
The instructions are a bit old and before the libimobiledevice-glue library was around so you need to add that.

Last time I did this (1 year ago), I found the need to use specific checkin for libplist and not the head as previously described.

Libplist checkout a9e34bd29ae9dcdae55bdf5fb8a23c9b1c02eee9

It may be different now. If I get a chance to try it again I’ll update the tutorial

cl532257933, 2023-08-16:
> I have the same issue @cl532257933.
>
> I followed https://www.numerousnetworks.co.uk/guides/untether-ios-step-by-step/ (I had to make some changes to the code because there were some duplication). [...]

yes,  I installed dependencies mixed of `apt-get`  and manual compiling and installing.I also modify the code to resolve repeat definiton  as follows:
-enum plist_format_t {
-       PLIST_FORMAT_XML,
-       PLIST_FORMAT_BINARY
+enum plist_format_t1 {
+       PLIST_FORMAT_XML1,
+       PLIST_FORMAT_BINARY1

-int plist_write_to_filename(plist_t plist, const char *filename, enum plist_format_t format);
+int plist_write_to_filename(plist_t plist, const char *filename, enum plist_format_t1 format);

cl532257933, 2023-08-16:
> The instructions are a bit old and before the libimobiledevice-glue library was around so you need to add that. Last time I did this (1 year ago), I found the need to use specific checkin for [...]

I didn't install  libimobiledevice-glue, I will try it again by your guidance

sepsky, 2023-08-16:
Thanks @cl532257933, I did something similar as well, I think one of them isn't actually used anywhere (I think it's the one in Util.c?) so I commented it out. 

I tried the same method (mix of apt-get and manual), as well as trying to build everything from repos, but that also failed with the exact same error. 

Also many thanks @bentumbler for the context and the commit you shared. Totally, I know it's been a while, and I encountered the failure for `libimobiledevice-glue` too. I fixed it by installing it from the repo. Not sure if this is relevant, but I built from `master`, if you used / want me to try a particular commit for that as well, please let me know.

I will try the commit you suggested, will rebuild and update with the outcome 👍

sepsky, 2023-08-16:
Hi @cl532257933 so I followed the script you have, and everything was installed ok (there were some additional requirements, such as `libgeneral` and `libimobiledevice-glue`, but apart from those it installed fine). 
However, I still get the same error as before from usbmuxd. Interestingly, I can see my devices when I run `avahi-discover` (you can install it using `sudo sudo apt-get install -y avahi-discover`) 

I've found a dockerfile, I've not tried it but I'll try it and will update the results as well as the dockerfile, in case you want to try it as well.

Hi @bentumbler so I tried the commit you provided, and had the same issue. I checked the code in `master` and it's got the changes in that commit as well. I think someone has already merged it into the `master`. Based on your experience, do you think it could be the new formatter we have to pass in to`plist_from_memory` method? I've tried PLIST_FORMAT_PLIST, PLIST_FORMAT_JSON and PLIST_FORMAT_NONE, and neither works (same error).

cl532257933, 2023-08-27:
@sepsky
I have resolved this issue #88 successfully，Execute command line “ifconfig eth0 -allmulti” to make ubuntu receive Multicast message. Maybe the  issue #88 could be closed.

dmitrym0, 2023-09-15:
> @sepsky
> I have resolved this issue #88 successfully，Execute command line “ifconfig eth0 -allmulti” to make ubuntu receive Multicast message.@fosple Maybe the issue #88 could be closed.

Did you install this on host or used a docker container? What version are you running?

```
ideviceinfo --version
ideviceinfo 1.3.0-164-gb5ce444
```

leodeve, 2024-04-03:
Any news?
I still cannot find my iPhone via network on my Linux (Linux Mint 21.3). It works well in USB.
I use the following docker container:
```dockerfile
FROM debian:11-slim

RUN apt-get update && apt-get install -y build-essential pkg-config checkinstall git autoconf automake

RUN apt-get install -y libtool-bin libssl-dev

RUN apt-get install -y libusb-1.0.0-dev libavahi-client-dev

ENV INSTALL_PATH /src
RUN mkdir -p $INSTALL_PATH
WORKDIR $INSTALL_PATH

RUN git clone https://github.com/libimobiledevice/libplist.git \
  && cd /src/libplist \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/libimobiledevice/libimobiledevice-glue.git \
  && cd /src/libimobiledevice-glue \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/tihmstar/libgeneral.git \
  && cd /src/libgeneral \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/libimobiledevice/libusbmuxd.git \
  && cd /src/libusbmuxd \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/libimobiledevice/libimobiledevice.git \
  && cd /src/libimobiledevice \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/tihmstar/usbmuxd2.git \
  && cd /src/usbmuxd2 \
  && apt-get install -y clang \
  && ./autogen.sh \
  && ./configure CC=clang CXX=clang++ \
  && make \
  && make install

RUN apt-get -y install ideviceinstaller
```

I followed this link to compile usbmuxd2: https://github.com/tihmstar/usbmuxd2/issues/31
I ran my container with the following commands:
```bash
docker build -t libimobiledevice-image .
docker run -it -v /var/run:/var/run --network host libimobiledevice-image
```

I ran the following commands in my container:
```bash
usbmuxd -v
[09:07:50.612][3] starting usbmuxd2 version: 0.61-2ce399ddbacb110bd5a83a6b8232d42c9a9b6e84-RELEASE
[09:07:50.614][2] Failed to get doPreflight! setting it to default val
[09:07:50.615][2] sysconf_set_value: Reading SystemConfiguration failed! Regenerating!
[09:07:50.616][2] Failed to get enableWifiDeviceManager! setting it to default val
[09:07:50.617][2] Failed to get enableUSBDeviceManager! setting it to default val
[09:07:50.622][3] Loaded config
[09:07:50.622][3] starting usbmuxd2 version: 0.61-2ce399ddbacb110bd5a83a6b8232d42c9a9b6e84-RELEASE
[09:07:50.622][0] Another instance is already running (pid 0). exiting.
[09:07:50.623][4] main reached cleanup
[09:07:50.623][4] done!
```
```bash
ideviceinfo --version
ideviceinfo 1.3.0-219-g3969bc9
```
```bash
idevicepair pair
SUCCESS: Paired with device 00008020-000815E82E06002E
```
```bash
idevicepair pair --wireless
No device found.
```
```bash
idevice_id
00008020-000815E82E06002E (USB)
```
```bash
idevice_id -n
#no result
```

I tried the dockerfile of @cemizm but it doesn't work for me.
I also tried to use the multicast command of @cl532257933 but it doesn't work either.
I tried to connect my iPhone and PC to the same network.
I also tried to set up a wireless hotspot to share my PC connection to my iPhone.
Any help will be appreciated.

bentumbler, 2024-04-03:
What wireless network is this?
mDNS is required to be supported and than can be blocked on many enterprise networks although it usually is enabled on a home network

Can you get Avahi to find the iPhone?

I followed this link to compile usbmuxd2: tihmstar/usbmuxd2#31<https://github.com/tihmstar/usbmuxd2/issues/31>

leodeve, 2024-04-05:
Thank you for your answer.
I tried with an entreprise network and a home network.
To run avahi in my container, I added the line `RUN apt-get install -y avahi-utils` to my Dockerfile. I also ran my container with the `--privileged` option.

I ran the following command in my container at home:
```bash
avahi-resolve-host-name iPhone.local
iPhone.local	192.168.1.15
```

I added the line `RUN apt-get install -y systemd` to my Dockerfile to use the resolvectl command. With the `resolvectl status` command, I can see that the mDNS option is not enabled (-mDNS) :
```bash
Link 3 (wlp2s0)
    Current Scopes: DNS
         Protocols: +DefaultRoute +LLMNR -mDNS -DNSOverTLS DNSSEC=no/unsupported
Current DNS Server: 2a01:cb05:83d1:be00:3ab5:c9ff:fe0c:9480
       DNS Servers: 192.168.1.1 2a01:cb05:83d1:be00:3ab5:c9ff:fe0c:9480
        DNS Domain: home
```

I can enable it with the `resolvectl mdns wlp2s0 yes` command. However, I still can't find my iPhone via network with libimobiledevice in my container.

leodeve, 2024-04-11:
I kept digging into my problem. I ran the commands from my dockerfile directly in a real debian 11 and it worked. I connected my iPhone and my PC to the same network. Then, I could see my device with the `idevice_id -n` command. I didn't need to run the `usbmuxd` command. Consequently, the problem seems to come from docker. Do you have any idea why it works on a real debian 11 and not on docker with a debian 11 image?

mexmer, 2024-04-12:
> I kept digging into my problem. I ran the commands from my dockerfile directly in a real debian 11 and it worked. I connected my iPhone and my PC to the same network. Then, I could see my device [...]

to discover devices over network, program needs to send multicast (bonjour discovery) to network, this is not possible from docker, because docker doesn't have direct access to your network. you can imagine docker image to be computer behind router, that does NAT (in this case router is your computer).

full virtualization (like vmware or proxmox), can have access to network trough virtual interface, but containers do not have direct access to network, they are always proxied trough docker service.

cemizm, 2024-04-12:
> to discover devices over network, program needs to send multicast (bonjour discovery) to network, this is not possible from docker, because docker doesn't have direct access to your network. you can [...]

or you just use host network mode for your container:

https://github.com/libimobiledevice/libusbmuxd/issues/88#issuecomment-1003709212

mexmer, 2024-04-12:
> I am already using it. Here is the command I use to run my container:
> 
> ```shell
> docker run -it -v /var/run:/var/run --network host --privileged libimobiledevice-image
> ```
> 
> By the way, I need to use the volume /var/run. Otherwise I cannot detect my iPhone via USB.

usbmuxd stores pid.run file into /var/run, if you have that folder inside your container, and is writeable by usbmuxd, it should be enough, you should check usbmuxd or usbmuxd2 manual, there should be all commandline parameters, and probly location of files it writes, or check in source code.
usbmuxd needs to write pid.run file, but also lockdown files from paired devices, and also log (if you have logging to file enabled)

leodeve, 2024-04-17:
I don't have a pid.run file into /var/run in my container and in my host.
I checked the usbmuxd2 source code. I can only see the following files: /var/run/usbmuxd.pid, /var/run/usbmuxd and /var/lib/lockdown.
I continued my research. I ran my docker container in a real debian 11 that has usbmuxd2 installed. It worked. I could see my device with the `idevice_id -n` command. It seems that the container needs to be run on a host with the usbmuxd2 service running.

mexmer, 2024-04-17:
> I don't have a pid.run file into /var/run in my container and in my host. I checked the usbmuxd2 source code. I can only see the following files: /var/run/usbmuxd.pid, /var/run/usbmuxd and [...]

by pid.run file i did mean usbmuxd.pid, i'm lazy to search code for exact name 😸 
as for what you say regarding usbmuxd, i thought you are trying to run usbmuxd2 inside container. of course usbmud2 running directly on computer will see all mdns and bonjour devices.

now i'm honestly lost in what you are actually trying to do.

Blackspell01, 2024-08-21:
I can see my iPhone with idevice_id -n but not connect to it. It says:

```
ERROR: [SRP] Pairing is blocked for another 5 seconds
ERROR: Pairing with device 00008120-00**********0201E failed.

```
Does someone know what the problem could be?

Blackspell01, 2024-09-11:
Has someone got working now?

leodeve, 2024-09-26:
Hello,
Can you specify your question?
I can find my iPhone via network (idevice_id -n) if I install the libimobiledevice tools directly on my PC.
I also wanted to find my iPhone via network with docker. It works only if the usbmuxd2 service is running on the PC. What are you trying to do?

Blackspell01, 2024-09-27:
I finally got it working inside docker. Inside my local network I can backup my iPhone wirelessly on my server running docker, exactly what I wanted. 

The only thing I haven't found out yet is how to backup the iPhone if it's not in the local network. I thought I could just connect the Phone over VPN and backup from somewhere else.

NN708, 2024-09-29:
> I finally got it working inside docker. Inside my local network I can backup my iPhone wirelessly on my server running docker, exactly what I wanted.

@Blackspell01 That's a great accomplishment! Would you consider sharing your Docker image?

Blackspell01, 2024-10-08:
Yes of course, here you are:

`FROM debian:11-slim

RUN apt-get update && apt-get upgrade -y
RUN apt-get install -y build-essential pkg-config checkinstall git autoconf automake 
RUN apt-get install -y libtool-bin libssl-dev libcurl4-openssl-dev
RUN apt-get install -y libusb-1.0.0-dev libavahi-client-dev avahi-daemon avahi-utils

ENV INSTALL_PATH /src
RUN mkdir -p $INSTALL_PATH
WORKDIR $INSTALL_PATH

RUN git clone https://github.com/libimobiledevice/libplist.git \
  && cd /src/libplist \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/libimobiledevice/libimobiledevice-glue.git \
  && cd /src/libimobiledevice-glue \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/libimobiledevice/libtatsu.git \
  && cd /src/libtatsu \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/libimobiledevice/libusbmuxd.git \
  && cd /src/libusbmuxd \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/libimobiledevice/libimobiledevice.git \
  && cd /src/libimobiledevice \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/tihmstar/libgeneral.git \
  && cd /src/libgeneral \
  && ./autogen.sh \
  && make \
  && make install

RUN git clone https://github.com/tihmstar/usbmuxd2.git \
  && cd /src/usbmuxd2 \
  && apt-get install -y clang \
  && ./autogen.sh \
  && ./configure CC=clang CXX=clang++ \
  && make \
  && make install
`

The only problem is, that my plan was to be able to perform Backups from outside of the network. Unfortunately, the iPhone doesn't support mDNS over VPN apparently. In case someone knows more, I would be interested!

dmitrym0, 2024-10-08:
That's fantastic thanks @Blackspell01. I will give this a go. Related question: has anyone had an opportunity to restore from one of these backups? Did it work?

fosple, 2025-01-16:
@Blackspell01 Thanks for sharing the docker image 👍 Can you also share the commands you run to start the docker container, to pair your ios device and backup it? So we have a full mini tutorial :) Would be awesome.

fosple, 2025-01-21:
@Blackspell01 I already have figured out how to build everything and get it running. 

I also created a fork of usbmuxd2 from tihmstar which supports direct network connection via ip addresses (I added a `--connect <IP>` parameter). This should allow you to use the software also in a VPN environment. Fork: https://github.com/fosple/usbmuxd2.

closed 2025-01-21 by fosple

Blackspell01, 2025-01-22:
Damn, this is absolutely amazing! Can't wait to try this out tomorrow!
Thank you man

fosple, 2025-01-22:
@Blackspell01 Just did some testing. VPN seems to work if your iOS device is on a WIFI network and connected to the VPN. It seems not to work, if your iOS device is on a cellular connection and connected to the VPN. The iOS device needs to respond on port `62078` for the connection to usbmuxd to work. You can test if your iOS device is able to connect when port 62078 is open, e.g. `nc -zv 192.168.1.10 62078`.