libimobiledevice/libusbmuxd #92: usbmuxd_subscribe call back didn't call

piaojin, 2020-05-16:
I try to call **usbmuxd_subscrib**：
int res = usbmuxd_subscribe(usbmux_event_cb, NULL);
And res = 0, but won't call **usbmux_event_cb**. '**usbmuxd_events_subscribe**' API has the same issue.

static void usbmux_event_cb(const usbmuxd_event_t *event, void *user_data) {

}

Xcode 11.4
MacOS 10.15.4

piaojin, 2020-05-16:
I found "device_monitor" didn't call. The debug code is "USBMUXD_API int usbmuxd_events_subscribe(usbmuxd_subscription_context_t *ctx, usbmuxd_event_cb_t callback, void *user_data)" line "int res = thread_new(&devmon, device_monitor, NULL);".
[image]

piaojin, 2020-05-16:
I have updated libimobiledevice, libplist, libusbmuxd, openssl and usbmuxd to the lastest version.

nikias, 2020-05-16:
What are you doing after calling usbmuxd_subscribe? Remember this runs in a new thread so if you don't wait in the main thread (or the thread you call usbmuxd_subscribe from) it won't get a chance to call the callback function.

piaojin, 2020-05-16:
@nikias  I call usbmuxd_subscribe  in the main thread. And I just updated libimobiledevice, libplist, libusbmuxd, openssl and usbmuxd to the lastest version.

piaojin, 2020-05-16:
@nikias This is MacApp demo. Call usbmuxd_subscribe in applicationDidFinishLaunching. And the callback function "usbmux_event_cb" didn't be called.
[MacDemo.zip](https://github.com/libimobiledevice/libusbmuxd/files/4638704/MacDemo.zip)

nikias, 2020-05-16:
Hmm maybe it can't connect to `/var/run/usbmuxd` socket file? Also you are using static libs which is not really what LGPL license was meant for.

nikias, 2020-05-16:
Oh actually you said `device_monitor` isn't called. Weird. Can you single step through `usbmuxd_events_subscribe` and see if it actually tries to call `thread_new` at all?

piaojin, 2020-05-17:
@nikias I single-step debugging through ```usbmuxd_events_subscribe``` and ```thread_new``` was called. The res is 0.
[image]

nikias, 2020-05-17:
Ok now break inside device_monitor here:
https://github.com/libimobiledevice/libusbmuxd/blob/master/src/libusbmuxd.c#L1098
and see where it stops

piaojin, 2020-05-17:
I break inside the start line of ```device_monitor```, but the breakpoint didn't come to the start line of ```device_monitor```. ```device_monitor``` didn't be called. 

[image]

nikias, 2020-05-29:
Well I guess this is because you are linking in the libraries as static libraries. If you compile shared libraries and use those it will probably work.

piaojin, 2020-06-01:
> Well I guess this is because you are linking in the libraries as static libraries. If you compile shared libraries and use those it will probably work.

Thanks, I will have a try.

Glen-Bai, 2021-04-13:
@piaojin Did you solve this problem? I met it too

piaojin, 2021-05-02:
> @piaojin Did you solve this problem? I met it too

Sorry No.

nikias, 2023-04-22:
See my assumption above why this doesn't work. I know the code works if done properly, it's being used in a lot of projects, for example in checkra1n app.

closed 2023-04-22 by nikias