libimobiledevice/libusbmuxd #93: segfault when device_udid is not specified

EthanArbuckle, 2020-05-29:
Hello, i'm getting a consistent crash on Catalina (10.15.4). It only happens when UDID is not specified. 

I'm on commit 910166b3d67652abb54dece7e1cacdc9dcfdb659, from yesterday. 

Crashlog:  https://ghostbin.co/paste/u482g

this is the free() causing my crash - https://github.com/libimobiledevice/libusbmuxd/blob/master/tools/iproxy.c#L414

```
user@users-imac-pro ~ % iproxy 2222 22
waiting for connection
accepted connection, fd = 4
zsh: segmentation fault  iproxy 2222 22
```

```
user@users-imac-pro ~ % iproxy 2222 22 -u f039c2b5af6bb1a360936352b80d5c073e090617
waiting for connection
accepted connection, fd = 4
waiting for connection
Requesting connecion to USB device handle 4 (serial: f039c2b5af6bb1a360936352b80d5c073e090617), port 22
```

nikias, 2020-05-29:
My bad. I fixed it with commit d79d1908eec1eee9431cee2c27680182a4db6b3b.

EthanArbuckle, 2020-05-29:
Thanks Nikias

nikias, 2020-05-29:
You're welcome :)