libimobiledevice/libusbmuxd #95: compilation error on the latest version of libusbmuxd

iFred09, 2020-06-17:
Hello,

I'm trying to compile the latest version of libusbmuxd and I have this error, and I never had this error before (i successfully compiled this one month ago).
Same error when I compile with the version in the "Release" tab.

[image]

Thank's!

elekven-m, 2020-06-22:
I get this error also. Building using gcc 8.1.0 under mingw.

Orbif, 2020-06-26:
Looks like freeifaddrs and getifaddrs aren't available on mingw.

I found out [some implementation of theses functions](https://github.com/tumagonx/pygi-mingw-patches/blob/master/aravis-0.3.x.patch) exist for [PyGObject](https://sourceforge.net/projects/pygobjectwin32/) project on Windows.

I created a [patch adding these functions](https://github.com/Orbif/libimobiledevice-patchs/blob/master/add-windows-compatibility.patch) to libimobiledevice. With this patch, the compilation works fine, but maybe there is a better way to do this on Windows ?

iFred09, 2020-06-26:
hi @Orbif I want to know how to use your patch. Thx !

brendonjkding, 2020-06-27:
> hi @Orbif I want to know how to use your patch. Thx !

1. Download libimobiledevice-socket-mingw-compatibility.patch and move it to libimobiledevice folder
2. `git apply libimobiledevice-socket-mingw-compatibility.patch`

iFred09, 2020-06-27:
it works fine for libusbmuxd and libimobiledevice, thank's !

haen7, 2020-06-28:
I built libusbmuxd and libimobiledevice using @Orbif's  patch. Now `idevice_id` shows network devices but `ideviceinfo -n` gives ERROR.
[image]

bricedenice59, 2020-08-25:
The patch quoted above does not resolve compilation issues I have on Windows.
When could we have a working version that compiles ?
[image]

XanaduAPI, 2020-09-11:
Hello, can this patch connect to network devices? I tried to delete the IPv6 code, but it showed when I connected to a network device LOCKDOWN_E_MUX_ERROR

XanaduAPI, 2020-09-11:
I'm sorry, I'm using a translation tool

XanaduAPI, 2020-09-12:
@FunkyM 

Can you ask when you plan to fix the version compilation of MinGW?

dkw72n, 2020-09-29:
> The patch quoted above does not resolve compilation issues I have on Windows.
> When could we have a working version that compiles ?
> [image]

add `-D_WIN32_WINNT=0x600` to `CFLAGS` if you don't need your program to run below Windows XP.

see: http://www.mingw.org/wiki/Use_more_recent_defined_functions for more information

XanaduAPI, 2020-09-29:
@dkw72n 
I tried, but it didn't help.
lack getifaddrs 、freeifaddrs and IFF_RUNNING

Maybe the author didn't consider supporting Win32

dkw72n, 2020-10-04:
> @dkw72n
> I tried, but it didn't help.
> lack getifaddrs 、freeifaddrs and IFF_RUNNING
> 
> Maybe the author didn't consider supporting Win32

You also need the patch mentioned [above](https://github.com/libimobiledevice/libusbmuxd/issues/95#issuecomment-650066193) for those functions

nikias, 2020-10-12:
Hi, I created a patch that adds some of the functionality to compile it on win32. However while it compiles, I have no idea if it actually works.
https://gist.github.com/nikias/98e1723c7dcab771ee6627267133b28f

nikias, 2020-10-12:
Oh and I am late to the game and didn't see @Orbif's patch 🤦

mexmer, 2020-10-16:
when i have time, i will give some pull request for visual studio compilation. just need to check, if i didn't break linux compilation or msys

XanaduAPI, 2020-10-22:
To  libusbmuxd libimobiledevice patch up https://github.com/Orbif/libimobiledevice-patchs/blob/master/libimobiledevice-socket-mingw-compatibility.patch
Also  idevice.c >>
 	device->udid = strdup(muxdev->udid);
	device->mux_id = muxdev->handle;
	device->version = 0;
	switch (muxdev->conn_type) {
	case CONNECTION_TYPE_USB:
		device->conn_type = CONNECTION_USBMUXD;
		device->conn_data = NULL;
		break;
	case CONNECTION_TYPE_NETWORK:
		device->conn_type = CONNECTION_NETWORK;
		size_t len = ((uint8_t*)muxdev->conn_data)[0];
		device->conn_data = malloc(len);
		memcpy(device->conn_data, muxdev->conn_data, len);
		device->conn_type = CONNECTION_USBMUXD;    //add
		device->conn_data = NULL;  //add
		break;
	default:
		device->conn_type = 0;
		device->conn_data = NULL;
		break;
	}
	return device;

The WiFi device can be successfully connected.
Here is the execution result:

D:\Download\Build>ideviceinfo.exe -u 2df9dfdd5c8d4fb91283b0a8e3ed00034dd3b94d -n
---- device info plist start -----
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
        <key>ConnectionType</key>
        <string>Network</string>
        <key>DeviceID</key>
        <integer>88</integer>
        <key>EscapedFullServiceName</key>
        <string>98:00:c6:cd:db:2f@fe80::9a00:c6ff:fecd:db2f._apple-mobdev2._tcp.local.</string>
        <key>InterfaceIndex</key>
        <integer>8</integer>
        <key>NetworkAddress</key>
        <data>
        FwAAAAAAAAD+gAAAAAAAABTjbF+XozfNCAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
        AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
        AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
        </data>
        <key>SerialNumber</key>
        <string>2df9dfdd5c8d4fb91283b0a8e3ed00034dd3b94d</string>
</dict>
</plist>

---- device info plist end -----
NetworkAddress [Length:128] [0x00] : _ _ _ _ _ _ _ _€_ _ _ _ _ _ __鉥l___梍7_蚠_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
NetworkAddress [Length:128] [0x00] : 23_0_0_0_0_0_0_0_-2_-128_0_0_0_0_0_0_20_-29_108_95_-105_-93_55_-51_8_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_0_
NetworkAddress [Length:128] [0x00] : 17_00_00_00_00_00_00_00_FFFFFFFE_FFFFFF80_00_00_00_00_00_00_14_FFFFFFE3_6C_5F_FFFFFF97_FFFFFFA3_37_FFFFFFCD_08_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_00_
NetworkAddress INT32 : 23
NetworkAddress INT64 : 23

15:28:11 lockdown.c:651 lockdownd_client_new(): device udid: 2df9dfdd5c8d4fb91283b0a8e3ed00034dd3b94d
15:28:11 lockdown.c:380 lockdownd_query_type(): called
15:28:11 property_list_service.c:132 internal_plist_send(): sending 284 bytes
15:28:11 service.c:144 service_send(): sending 4 bytes
15:28:12 idevice.c:623 idevice_connection_send(): internal_connection_send 4, sent 4
15:28:12 service.c:144 service_send(): sending 284 bytes
15:28:12 idevice.c:623 idevice_connection_send(): internal_connection_send 284, sent 284
15:28:12 property_list_service.c:137 internal_plist_send(): sent 284 bytes
15:28:12 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
15:28:12 property_list_service.c:211 internal_plist_receive_timeout(): 297 bytes following
[... 119 lines cut ...]
WirelessBoardSerialNumber: 77EC8F00E9
kCTPostponementInfoPRLName: 0
kCTPostponementInfoServiceProvisioningState: false
kCTPostponementStatus: kCTPostponementStatusActivated
15:28:13 lockdown.c:262 lockdownd_stop_session(): stopping session 976C1876-649F-4D29-8E41-1126574D91A7
15:28:13 property_list_service.c:132 internal_plist_send(): sending 363 bytes
15:28:13 service.c:144 service_send(): sending 4 bytes
15:28:13 idevice.c:606 idevice_connection_send(): SSL_write 4, sent 4
15:28:13 service.c:144 service_send(): sending 363 bytes
15:28:13 idevice.c:606 idevice_connection_send(): SSL_write 363, sent 363
15:28:13 property_list_service.c:137 internal_plist_send(): sent 363 bytes
15:28:13 idevice.c:732 idevice_connection_receive_timeout(): SSL_read 4, received 4
15:28:13 property_list_service.c:205 internal_plist_receive_timeout(): initial read=4
15:28:13 property_list_service.c:211 internal_plist_receive_timeout(): 238 bytes following
15:28:13 idevice.c:732 idevice_connection_receive_timeout(): SSL_read 238, received 238
15:28:13 property_list_service.c:224 internal_plist_receive_timeout(): received 238 bytes
15:28:13 lockdown.c:278 lockdownd_stop_session(): success
15:28:13 idevice.c:1216 idevice_connection_disable_bypass_ssl(): Skipping bidirectional SSL shutdown. SSL error code: 5

15:28:13 idevice.c:1231 idevice_connection_disable_bypass_ssl(): SSL mode disabled

D:\Download\Build>

I suspect that the problem is that libusbmuxd tries to connect WiFi devices through socket. However, due to the problem of IP address resolution under windows, I changed it to USB mode, but I did not connect USB device (refer to the previous compiler version)
The last compiled version can also be connected to WiFi devices

haen7, 2020-10-22:
after applying @XanaduAPI's patch everything seems to be working fine. I just backed up my iphone over wifi on windows 10 using idevicebackup2.

nikias, 2023-06-27:
The original issue should be resolved.