libimobiledevice/libusbmuxd #98: new port format not backwards compatible

paradite, 2020-06-30:
the commit that changed the format https://github.com/libimobiledevice/libusbmuxd/commit/3ae1cbbfb3b21af9ab1bc749b6a2681a3a3334ec was not backwards compatible as it claimed.
old format:
```
$ iproxy
usage: iproxy LOCAL_TCP_PORT DEVICE_TCP_PORT [UDID]
```
this works correctly:
```
$ iproxy 8001 8001 7af24cbfe52bdf7be67b20d61409297521a76458
waiting for connection
```

new format:
```
Usage: iproxy [OPTIONS] LOCAL_PORT:DEVICE_PORT [LOCAL_PORT2:DEVICE_PORT2 ...]
```

this breaks and throws error:

```
$ iproxy 8001 8001 7af24cbfe52bdf7be67b20d61409297521a76458
Invalid listen port specified in argument '8001'!
```

this is tested on macOS with `brew install usbmuxd`.

paradite, 2020-06-30:
for anyone looking for a temporary solution:
```bash
brew unlink usbmuxd
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/afce634843e7c5c4ec9be88479d1f6631ba9ac4a/Formula/libusbmuxd.rb
brew switch usbmuxd 2.0.1

# if there is libplist error:
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/d74b26325915bdf33284af7c4bf0993e99616919/Formula/libplist.rb
```

bottotl, 2020-09-13:
iproxy [OPTIONS] LOCAL_PORT:DEVICE_PORT [LOCAL_PORT2:DEVICE_PORT2 ...] 变成这样了直接用

nikias, 2020-09-13:
The old format works when you don't pass the udid as 3rd argument (could still be passed as -u <UDID>). But just use the new format, I don't see the point in reverting to an old version 😆🤷‍♂️

closed 2020-09-13 by nikias