libimobiledevice/libusbmuxd #100: How to build libusbmuxd on a Windows machine?

nSneerfulBike, 2020-10-27:
I know this might sound like a stupid question, but I am really struggling to understand how to build this library on Windows 10. I have never done something like this before and I am not finding any tutorial about it. I have also tried with the win32 variant of this library from [here](https://github.com/libimobiledevice-win32/usbmuxd), but it gives errors over errors. I have no idea what to do either with MinGW or CMake, so can anybody help me with this?

XanaduAPI, 2020-11-02:
@nSneerfulBike  你应该使用 Mingw32 [http://repo.msys2.org/distrib/i686/msys2-i686-20200517.exe] ，然后按照提示安装一些包。
pacman -S mingw-w64-i686-toolchain
pacman -S --needed base-devel msys2-devel
pacman -S openssl-devel
pacman -S libreadline-devel
pacman -S mingw-w64-i686-libusb
pacman -S mingw-w64-i686-libzip
pacman -S mingw-w64-i686-libxml2
pacman -S mingw-w64-i686-libssh2
pacman -S mingw-w64-i686-curl
pacman -S git

再构建 以下包
git clone https://github.com/libimobiledevice/libplist.git
./autogen.sh

git clone https://github.com/libimobiledevice/libusbmuxd.git
./autogen.sh

git clone https://github.com/libimobiledevice/libimobiledevice.git
./autogen.sh

XanaduAPI, 2020-11-02:
@nSneerfulBike   如果编译方面有错误，请参照[https://github.com/libimobiledevice/libusbmuxd/issues/95]compilation error on the latest version of libusbmuxd on windows

nSneerfulBike, 2020-11-02:
> @nSneerfulBike 你应该使用 Mingw32 [http://repo.msys2.org/distrib/i686/msys2-i686-20200517.exe] ，然后按照提示安装一些包。
> pacman -S mingw-w64-i686-toolchain
> pacman -S --needed base-devel msys2-devel
> pacman -S [...]

So autogen.sh works on Windows too?

XanaduAPI, 2020-11-02:
@nSneerfulBike 
[image]
是的，但是依赖于 Mingw32/64

nSneerfulBike, 2020-11-02:
> @nSneerfulBike
> [image]
> 是的，但是依赖于 Mingw32/64

Thanks a lot for replying, I'll let you know tomorrow/later, it's late in my country right now

XanaduAPI, 2020-11-02:
@nSneerfulBike  晚安，我这里才中午12点

nSneerfulBike, 2020-11-02:
> @nSneerfulBike 你应该使用 Mingw32 [http://repo.msys2.org/distrib/i686/msys2-i686-20200517.exe] ，然后按照提示安装一些包。
> pacman -S mingw-w64-i686-toolchain
> pacman -S --needed base-devel msys2-devel
> pacman -S [...]

Sorry but I am totally unable to compile using `./autogen.sh`. I get the following error:

checking for cython... /mingw64/bin/cython
checking for Cython version... 0.29.21
configure: Cython executable is '/mingw64/bin/cython'
checking for python3.8... (cached) /mingw64/bin/python
checking for a version of Python >= '2.1.0'... yes
checking for the distutils Python package... yes
checking for Python include path... -IC:/msys32/mingw64/include/python3.8
checking for Python library path... -LC:/msys32/mingw64/lib -lpython3.8
checking for Python site-packages path... C:/msys32/mingw64/lib/python3.8/site-packages
checking python extra libraries... -lm -lversion -lshlwapi -lm
checking python extra linking flags...
checking consistency of all components of python development environment... no
configure: error: in `/c/Users/vincy/Desktop/libimobiledeviceFolder/libplist':
configure: error:
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LIBS environment variable.
  Example: ./configure LIBS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================

See `config.log' for more details

XanaduAPI, 2020-11-03:
@nSneerfulBike   你可能需要安装 Python 的对应版本。
或许你可以尝试安装以下工具，可能对你会有所帮助。
[image]

nikias, 2020-11-03:
@nSneerfulBike you can just pass `--without-cython` on the command line if you don't need the python bindings.

nSneerfulBike, 2020-11-03:
> @nSneerfulBike 你可能需要安装 Python 的对应版本。
> 或许你可以尝试安装以下工具，可能对你会有所帮助。
> [image]
> 
> $ pacman -Ss python | grep 已安装
> mingw32/mingw-w64-i686-python 3.8.6-3 [已安装]
> mingw32/mingw-w64-i686-python2 2.7.18-1 [已安装]
> mingw32/mingw-w64-i686-python2-setuptools 44.1.1-1 [已安装]
> msys/python 3.8.2-1 [已安装]
> msys/python2 2.7.18-1 [已安装]
> msys/scons 3.1.2-4 (base-devel) [已安装]

Okay I now have all those things installed and actually discovered that the problem was me using MSYS2 MinGW 64-bit instead of 32-bit. Now I have done:
`./autogen.sh`
`make`
`make install`
with libplist, but it doesn't work with libusbmuxd. `./autogen.sh` works, but then when I type in `make`, I get the following:

$ make
make  all-recursive
make[1]: Entering directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libusbmuxd'
Making all in common
make[2]: Entering directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libusbmuxd/common'
  CC       socket.lo
socket.c: In function '_sockaddr_in6_scope_id':
socket.c:394:6: warning: implicit declaration of function 'getifaddrs' [-Wimplicit-function-declaration]
  394 |  if (getifaddrs(&ifaddr) == -1) {
      |      ^~~~~~~~~~
socket.c:400:43: error: invalid use of undefined type 'struct ifaddrs'
  400 |  for (ifa = ifaddr; ifa != NULL; ifa = ifa->ifa_next) {
      |                                           ^~
socket.c:402:10: error: invalid use of undefined type 'struct ifaddrs'
  402 |   if (ifa->ifa_addr == NULL) {
      |          ^~
socket.c:407:10: error: invalid use of undefined type 'struct ifaddrs'
  407 |   if (ifa->ifa_addr->sa_family != AF_INET6) {
      |          ^~
socket.c:412:11: error: invalid use of undefined type 'struct ifaddrs'
  412 |   if ((ifa->ifa_flags & IFF_UP) == 0) {
      |           ^~
socket.c:417:11: error: invalid use of undefined type 'struct ifaddrs'
  417 |   if ((ifa->ifa_flags & IFF_RUNNING) == 0) {
      |           ^~
socket.c:417:25: error: 'IFF_RUNNING' undeclared (first use in this function)
  417 |   if ((ifa->ifa_flags & IFF_RUNNING) == 0) {
      |                         ^~~~~~~~~~~
socket.c:417:25: note: each undeclared identifier is reported only once for each function it appears in
socket.c:421:59: error: invalid use of undefined type 'struct ifaddrs'
  421 |   struct sockaddr_in6* addr_in = (struct sockaddr_in6*)ifa->ifa_addr;
      |                                                           ^~
socket.c:445:11: error: invalid use of undefined type 'struct ifaddrs'
  445 |   if ((ifa->ifa_flags & IFF_LOOPBACK) != 0) {
      |           ^~
socket.c:463:2: warning: implicit declaration of function 'freeifaddrs' [-Wimplicit-function-declaration]
  463 |  freeifaddrs(ifaddr);
      |  ^~~~~~~~~~~
make[2]: *** [Makefile:402: socket.lo] Error 1
make[2]: Leaving directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libusbmuxd/common'
make[1]: *** [Makefile:417: all-recursive] Error 1
make[1]: Leaving directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libusbmuxd'
make: *** [Makefile:349: all] Error 2

nSneerfulBike, 2020-11-03:
> @nSneerfulBike you can just pass `--without-cython` on the command line if you don't need the python bindings.

Thank you but I have now managed to get it working with this, now there's another problem

XanaduAPI, 2020-11-03:
@nSneerfulBike https://github.com//issues/95

XanaduAPI, 2020-11-03:
@nSneerfulBike  你应该看看我和其它人在 Windows 下编译的问题。现在已经解决了你编译出现的错误。

nSneerfulBike, 2020-11-03:
> @nSneerfulBike #95

The patch works for libusbmuxd but with libimobiledevice gives the following error when typing `make`:

$ make
make  all-recursive
make[1]: Entering directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice'
Making all in common
make[2]: Entering directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/common'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/common'
Making all in src
make[2]: Entering directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/src'
Making all in include
make[2]: Entering directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/include'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/include'
Making all in cython
make[2]: Entering directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/cython'
make  all-am
make[3]: Entering directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/cython'
  CCLD     imobiledevice.la
C:/msys32/mingw32/bin/../lib/gcc/i686-w64-mingw32/10.1.0/../../../../i686-w64-mingw32/bin/ld.exe: .libs/imobiledevice_la-imobiledevice.o: in function `_pyx_f_13imobiledevice_to_cstring_array':
C:\Users\vincy\Desktop\libimobiledeviceFolder\libimobiledevice\cython/imobiledevice.c:54374: undefined reference to `PyString_AsString'
collect2.exe: error: ld returned 1 exit status
make[3]: *** [Makefile:530: imobiledevice.la] Error 1
make[3]: Leaving directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/cython'
make[2]: *** [Makefile:460: all] Error 2
make[2]: Leaving directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice/cython'
make[1]: *** [Makefile:454: all-recursive] Error 1
make[1]: Leaving directory '/c/Users/vincy/Desktop/libimobiledeviceFolder/libimobiledevice'
make: *** [Makefile:384: all] Error 2

XanaduAPI, 2020-11-03:
@nSneerfulBike  这个补丁，你应该同时应用于 libusbmuxd 和 libimobiledevice，并且，最好是自己修改 源代码。
修改好后，你还要应用我 提到的，把 idevice.c 中的某一段修改为 USB设备。因为 libimobiledevice 在Windows下，连接不上WIFI设备。

XanaduAPI, 2020-11-03:
@nSneerfulBike  也许你可以使用我编译好的二进制文件。
我可以通过邮箱发送给你

nSneerfulBike, 2020-11-03:
> @nSneerfulBike 也许你可以使用我编译好的二进制文件。
> 我可以通过邮箱发送给你

You may help me with this release, but then you should send me the binaries for every release, so better to fix existing problems.

> @nSneerfulBike 这个补丁，你应该同时应用于 libusbmuxd 和 libimobiledevice，并且，最好是自己修改 源代码。
> 修改好后，你还要应用我 提到的，把 idevice.c 中的某一段修改为 USB设备。因为 libimobiledevice 在Windows下，连接不上WIFI设备。

What do you mean? I have tried to re-clone both libusbmuxd and libimobiledevice, apply the patch to both of them and only then run `./autogen.sh`, `make` and `make install` but still same problem: `make` gives the same error on libimobiledevice

XanaduAPI, 2020-11-03:
@nSneerfulBike 你确定是使用的 对应 x86 / x64 的库吗？我看到上面出现了 [_pyx_f_13imobiledevice_to_cstring_array': C:\Users\vincy\Desktop\libimobiledeviceFolder\libimobiledevice\cython/imobiledevice.c:54374: undefined reference to PyString_AsString'] 如果可以，能不能尝试下重新安装 Python 呢？或者建立一个虚拟机 从 libplist > libusbmuxd > libimobiledevice 重新构建呢，保证环境无污染。 当然了，如果你需要二进制文件，我可以给你我最新编译的。更早之前的我没有保存。

nSneerfulBike, 2020-11-03:
> @nSneerfulBike 你确定是使用的 对应 x86 / x64 的库吗？我看到上面出现了 [_pyx_f_13imobiledevice_to_cstring_array': C:\Users\vincy\Desktop\libimobiledeviceFolder\libimobiledevice\cython/imobiledevice.c:54374: undefined [...]

You made me install the i686 (x86 only) version, and actually I don't know how installing the x86_64 version could be useful since I am using the 32-bit version of MSYS2

XanaduAPI, 2020-11-03:
@nSneerfulBike 
[image]
i mean：You can confirm again that the dependent libraries have been installed.
Also, you can create a virtual machine to install from scratch,Make sure the environment is not polluted.

Python libraries are best installed as well.

Before, you said that you used the 64 bit environment, so I want you to make sure that you have not installed the dependency library with the wrong number of digits.

[The above belongs to machine translation]

nSneerfulBike, 2020-11-03:
> @nSneerfulBike
> [image]
> i mean：You can confirm again that the dependent libraries have been installed.
> Also, you can create a virtual machine to install from scratch,Make sure the environment is not [...]

And then when I have finished doing `make` and `make install` where can I find the compiled binaries?
Anyway you can keep writing in Chinese and I'll use a translator

nSneerfulBike, 2020-11-03:
I fixed the problem I had by replacing every `PyString_AsString` with `PyUnicode_AsUTF8` referring to [this](https://stackoverflow.com/questions/36530637/embedding-python-in-c-error-in-linking-undefined-reference-to-pystring-asstri) StackOverflow question, and it worked. I am now able to use the executables

XanaduAPI, 2020-11-04:
@nSneerfulBike  我担心的是，中文翻译为其它语言可能会改变一部分意思。一般二进制文件会在 [src/.libs] 或 [tools/.libs] 文件夹里面。或许你可以安装一个 Everything ，这会方便你查找文件。

nSneerfulBike, 2020-11-04:
> @nSneerfulBike 我担心的是，中文翻译为其它语言可能会改变一部分意思。一般二进制文件会在 [src/.libs] 或 [tools/.libs] 文件夹里面。或许你可以安装一个 Everything ，这会方便你查找文件。

Thanks for your reply but I have already found every file. The problem now is that idevicerestore does not work. The only "useful" thread for me is [this](https://github.com/libimobiledevice/idevicerestore/issues/311) one, but it still doesn't solve my problem. I just type
`idevicerestore -e -y "C:\Users\vincy\Desktop\a\iPhone_5.5_P3_14.1_18A8395_Restore.ipsw"`
and I get the following:

Found device in Recovery mode
Identified device as d111ap, iPhone9,4
Extracting BuildManifest from IPSW
Product Version: 14.1
Product Build: 18A8395 Major: 18
ERROR: Device is in an invalid state
Device supports Image4: false
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Checking IPSW for required components...
All required components found in IPSW
Using cached filesystem from 'C:\Users\vincy\Desktop\a\iPhone_5.5_P3_14.1_18A8395_Restore/038-74057-065.dmg'
ERROR: Device is in an invalid state
ERROR: Unable to find device ECID

XanaduAPI, 2020-11-04:
@nSneerfulBike  如果你使用 screenshotr_take_screenshot 、sbservices_get_home_screen_wallpaper_pngdata 等函数，你可能需要修改源文件，添加上 screenshotr_screenshot_image_free(char* _Picture)等函数，用来释放内存。
在Windows中，跨模块释放内存是不可取的。

XanaduAPI, 2020-11-04:
@nSneerfulBike  对此

> Thanks for your reply but I have already found every file. The problem now is that idevicerestore does not work. The only "useful" thread for me is [...]

对此我也没有更好的办法，或许你可以使用我编写的软件 http://www.shalou.net/home 尝试能否为你的设备进行刷机。

nSneerfulBike, 2020-11-04:
> @nSneerfulBike 对此
>
>
> 对此我也没有更好的办法，或许你可以使用我编写的软件 http://www.shalou.net/home 尝试能否为你的设备进行刷机。

Have you built it using imobiledevice?

XanaduAPI, 2020-11-04:
@nSneerfulBike  刷机部分并不是使用的 imobiledevice。

nSneerfulBike, 2020-11-04:
> @nSneerfulBike 或许你可以修改下参数。-i -u -C

I have only one iOS device connected so I don't put neither the -u nor the -i since it works the same way, still gives error. I have anyway run:
`idevicerestore -e -y -C "C:\Users\vincy\Desktop\b" "C:\Users\vincy\Desktop\a\iPhone_5.5_P3_14.1_18A8395_Restore.ipsw"`
and I get the same as before

nSneerfulBike, 2020-11-04:
> 建议提前把设备进入恢复模式，然后指定 ecid 进行刷机，这样可能成功率更高一些。

The device is already in recovery mode. When I start the command while it isn't it tells me to turn on usbmuxd, but it is on

XanaduAPI, 2020-11-04:
> I have only one iOS device connected so I don't put neither the -u nor the -i since it works the same way, still gives error. I have anyway run:
> `idevicerestore -e -y -C "C:\Users\vincy\Desktop\b" [...]

如果你指定了 -C，那么你只需要传入 固件解压后的目录就可以了，并不需要再传入 .ipsw 文件。

nSneerfulBike, 2020-11-04:
> 如果你指定了 -C，那么你只需要传入 固件解压后的目录就可以了，并不需要再传入 .ipsw 文件。

When I do that it tells me that the command is wrong

XanaduAPI, 2020-11-04:
> > 如果你指定了 -C，那么你只需要传入 固件解压后的目录就可以了，并不需要再传入 .ipsw 文件。
> 
> When I do that it tells me that the command is wrong

你需要提前解压固件，我就是这样做的。
而且，我并没有遇到你出现的  usbmuxd 错误，或许这个问题你可以 询问 作者。

nSneerfulBike, 2020-11-04:
> > > 如果你指定了 -C，那么你只需要传入 固件解压后的目录就可以了，并不需要再传入 .ipsw 文件。
> > 
> > 
> > When I do that it tells me that the command is wrong
> 
> 你需要提前解压固件，我就是这样做的。
> 而且，我并没有遇到你出现的 usbmuxd 错误，或许这个问题你可以 询问 作者。

C:\Users\vincy\Desktop\imobiledevice compiled>idevicerestore -e -y -C "C:\Users\vincy\Desktop\b" -i 0x001271e808f110a6
Usage: idevicerestore [OPTIONS] PATH

Restore IPSW firmware at PATH to an iOS device.

PATH can be a compressed .ipsw file or a directory containing all files
extracted from an IPSW.

OPTIONS:
  -i, --ecid ECID       Target specific device by its ECID
                        e.g. 0xaabb123456 (hex) or 1234567890 (decimal)
  -u, --udid UDID       Target specific device by its device UDID
                        NOTE: only works with devices in normal mode.
  -l, --latest          Use latest available firmware (with download on demand).
                        Before performing any action it will interactively ask
                        to select one of the currently signed firmware versions,
                        unless -y has been given too.
                        The PATH argument is ignored when using this option.
                        DO NOT USE if you need to preserve the baseband/unlock!
                        USE WITH CARE if you want to keep a jailbreakable
                        firmware!
  -e, --erase           Perform full restore instead of update, erasing all data
                        DO NOT USE if you want to preserve user data on the device!
  -y, --no-input        Non-interactive mode, do not ask for any input.
                        WARNING: This will disable certain checks/prompts that
                        are supposed to prevent DATA LOSS. Use with caution.
  -n, --no-action       Do not perform any restore action. If combined with -l
                        option the on-demand ipsw download is performed before
                        exiting.
  -h, --help            Prints this usage information
  -C, --cache-path DIR  Use specified directory for caching extracted or other
                        reused files.
  -d, --debug           Enable communication debugging
  -v, --version         Print version information

Advanced/experimental options:
  -c, --custom          Restore with a custom firmware (requires bootrom exploit)
  -s, --cydia           Use Cydia's signature service instead of Apple's
  -x, --exclude         Exclude nor/baseband upgrade
  -t, --shsh            Fetch TSS record and save to .shsh file, then exit
  -z, --no-restore      Do not restore and end after booting to the ramdisk
  -k, --keep-pers       Write personalized components to files for debugging
  -p, --pwn             Put device in pwned DFU mode and exit (limera1n devices)
  -P, --plain-progress  Print progress as plain step and progress
  -R, --restore-mode    Allow restoring from Restore mode
  -T, --ticket PATH     Use file at PATH to send as AP ticket

Homepage:    <https://libimobiledevice.org>
Bug Reports: <https://github.com/libimobiledevice/idevicerestore/issues>

Yes, I have extracted the .ipsw using WinRAR into "C:\Users\vincy\Desktop\b"

nSneerfulBike, 2020-11-04:
@XanaduAPI it is no different if I put the "-i 0xxx" before the "-C"

XanaduAPI, 2020-11-04:
> @XanaduAPI it is no different if I put the "-i 0xxx" before the "-C"

ECID或许不会需要 0x 开头。如果还有其它的问题，我想你应该去 [https://github.com/libimobiledevice/idevicerestore/issues] 找一找有没有解决方案

nSneerfulBike, 2020-11-04:
@XanaduAPI nevermind, it looks like it is working now. I have uninstalled all the Apple softwares I had on my PC and reinstalled everything after removing the iBoot driver, and it did neither give any error nor stop suddenly. Looks great for now

XanaduAPI, 2020-11-04:
@nSneerfulBike  如果的设备硬件正常并且与苹果网络的连接良好，一般不会出现问题。祝你顺利

nSneerfulBike, 2020-11-04:
> @nSneerfulBike 如果的设备硬件正常并且与苹果网络的连接良好，一般不会出现问题。祝你顺利

Thanks mate, but it looks like that the iBoot driver gives tons of problem. After removing it and reinstalling iTunes from scratch, it's gone pretty straight forward

nikias, 2020-11-04:
> C:\Users\vincy\Desktop\imobiledevice compiled>idevicerestore -e -y -C "C:\Users\vincy\Desktop\b" -i 0x001271e808f110a6
> Usage: idevicerestore [OPTIONS] PATH
>
> Restore IPSW firmware at PATH to an iOS [...]

You are not passing the ipsw file or path. `-C` is something different it is a cache folder location for extracted material. If you want to use an extracted ipsw folder, you just pass it as the PATH argument just as the help output says:
`PATH can be a compressed .ipsw file or a directory containing all files extracted from an IPSW.`

WuMoWen, 2020-11-20:
> > @nSneerfulBike你应该使用MINGW32 [http://repo.msys2.org/distrib/i686/msys2-i686-20200517.exe]，然后按照提示安装一些包。
> > 吃豆-S的mingw-W64-i686的工具链
> > 吃豆-S -需要基础开发msys2-开发
> > pacman -S openssl-开发
> > pacman -S libreadline-开发
> > pacman -S mingw-w64-i686-libusb 
> > pacman -S mingw-w64-i686-libzip 
> > pacman -S mingw-w64-i686- libxml2 
> > pacman -S mingw-w64-i686-libssh2 
> > pacman -S mingw-w64-i686-curl 
> > pacman -S git
> > 再构建以下包
> > git clone https://github.com/libimobiledevice/libplist.git 
> > ./autogen.sh
> > git clone https://github.com/libimobiledevice/libusbmuxd.git 
> > ./autogen.sh
> > git clone https://github.com/libimobiledevice/libimobiledevice.git 
> > ./autogen.sh
> 
> 抱歉，我完全无法使用进行编译`./autogen.sh`。我收到以下错误：
> 
> # 检查cython ... / mingw64 / bin / cython
> 检查Cython版本... 0.29.21
> 配置：Cython可执行文件为'/ mingw64 / bin / cython'
> 检查python3.8 ...（已缓存）/ mingw64 / bin / python
> 检查Python版本> ='2.1.0'...是，
> 检查distutils Python软件包...是，
> 检查Python包含路径... -IC：/msys32/mingw64/include/python3.8
> 检查Python库路径... -LC：/ msys32 / mingw64 / lib -lpython3.8
> 检查Python站点软件包路径... C：/msys32/mingw64/lib/python3.8/site-packages
> 检查python额外库... -lm -lversion -lshlwapi -lm
> 检查python额外的链接标志...
> 正在检查python开发环境的所有组件的一致性...否
> 配置：错误：在`/ c / Users / vincy / Desktop / libimobiledeviceFolder / libplist'中：
> 配置：错误：
> 无法将测试程序链接到Python。可能主Python库已
> 安装在某些非标准的库路径中。如果是这样，则
> 通过LIBS环境变量将其传递给配置。
> 示例：./configure LIBS =“-L / usr / non-standard-path / python / lib”
> # 错误！
> 您可能需要为发行版安装Python软件包的开发版本
> 。该软件包的确切名称在其中有所不同。
> 参见`config.log'了解更多细节

我出现了和你一样的问题，请问你是怎么解决的

XanaduAPI, 2020-11-20:
@WuMoWen 
你可以试一下
./autogen.sh --without-cython

WuMoWen, 2020-11-20:
@XanaduAPI 
好的 谢谢 方便加个QQ吗

WuMoWen, 2020-11-20:
@XanaduAPI 
好的 谢谢

zytzjx, 2020-12-29:
> Okay I now have all those things installed and actually discovered that the problem was me using MSYS2 MinGW 64-bit instead of 32-bit. Now I have done:
> `./autogen.sh`
> `make`
> `make install`
> with [...]

How to fix? I have the same problem.

XanaduAPI, 2020-12-29:
@zytzjx  作者好像已经更新修复这个问题，如果你要编译 libimobiledevice ，需要照着 libusbmuxd 里面的 Socket.h Socket.c 进行修改

XanaduAPI, 2020-12-29:
I think, is it time to close this issue?

fsinnes, 2021-07-22:
This is so fucking complicated, @nSneerfulBike can you send me a compile for the latest version?