libimobiledevice/libusbmuxd #103: iProxy terminal output event handler

ajschwieterman, 2020-11-24:
I'm trying to get the output of an iproxy command using the following code.
```
const { spawn } = require('child_process');
const cmd = spawn(`iproxy`, [`10101`, `10101`]);
cmd.stdout.on(`data`, (data) => {
    console.log("DATA!!!!");
});
```
When executing the command in the terminal I can see the following response in the terminal.
```
Creating listening port 10015 for device port 10015
waiting for connection
```
However, when using the above code, the `data` event isn't triggered in the child process.  Maybe I'm not using the correct event handler; I'm not sure what's happening.

ajschwieterman, 2020-11-24:
Maybe related to #79?

bartekpacia, 2022-08-10:
Today I encountered the same problem. Here's the minimum example of stdout not 

```bash
$ iproxy 8081:8081
Creating listening port 8081 for device port 8081
waiting for connection

# CTRL+C, this is okay
$ iproxy 8081:8081 1> test.txt 2>&1
# CTRL+C, let's see what was written
$ cat test.txt
# the file is empty - not expected
```

I was very annoyed by this, but a friend helped me and together we figured out how to make it work:

```
$ stdbuf -i0 -o0 -e0 iproxy 8081:8081 1> test.txt 2>&1
# CTRL+C
$ cat test.txt
Creating listening port 8081 for device port 8081
waiting for connection
```