libimobiledevice/libusbmuxd #105: Is there a way to forward just one device connection from usbmuxd

harrain, 2020-12-14:
When I use socat to forward /var/run/usbmuxd to tcp port:

socat TCP-LISTEN:1234,reuseaddr,fork UNIX-CONNECT:/var/run/usbmuxd

It will transfer all ios devices. Is there and way to transfer just one device ???

I want to share my ios device via internet, but I just want to share only 1 device for 1 client.

mexmer, 2020-12-14:
no, because you forwarding connection to USBMUXD, not to device.

usbmuxd forwards individual TCP sockets from device, but there is only 1 socket open on device all time, and that is lockdown service, other sockets will open, when you start service on device, and it will open at random port. so you will need to create new forward each time service socket is opened on device.

harrain, 2020-12-21:
So how to create new forward from LOCKDOWN service？appreciate

mexmer, 2020-12-21:
you need to write your own service, that will do it.