libimobiledevice/libusbmuxd #122: make failed for not found for architecture arm64

jagdish24, 2021-12-10:
libusbmuxd % make

System
Apple Macbook Pro 2021 M1

ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

/```
Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
  CC       libusbmuxd.lo
  CCLD     libusbmuxd-2.0.la
ld: warning: ignoring file /usr/local/lib/libplist-2.0.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /usr/local/Cellar/libplist/HEAD-cf7a3f3/lib/libplist-2.0.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_plist_array_get_item", referenced from:
      _usbmuxd_get_device_list in libusbmuxd.o
  "_plist_array_get_size", referenced from:
      _usbmuxd_get_device_list in libusbmuxd.o
  "_plist_copy", referenced from:
      _send_pair_record_packet in libusbmuxd.o
  "_plist_dict_get_item", referenced from:
      _usbmuxd_get_device_list in libusbmuxd.o
      _device_info_from_plist in libusbmuxd.o
      _receive_packet in libusbmuxd.o
      _usbmuxd_read_buid in libusbmuxd.o
      _usbmuxd_read_pair_record in libusbmuxd.o
      _create_plist_message in libusbmuxd.o
  "_plist_dict_set_item", referenced from:
      _usbmuxd_connect in libusbmuxd.o
      _usbmuxd_read_pair_record in libusbmuxd.o
      _send_pair_record_packet in libusbmuxd.o
      _usbmuxd_delete_pair_record in libusbmuxd.o
      _create_plist_message in libusbmuxd.o
  "_plist_free", referenced from:
      _device_monitor in libusbmuxd.o
      _usbmuxd_get_device_list in libusbmuxd.o
      _receive_packet in libusbmuxd.o
      _usbmuxd_connect in libusbmuxd.o
      _usbmuxd_read_buid in libusbmuxd.o
      _usbmuxd_read_pair_record in libusbmuxd.o
      _send_pair_record_packet in libusbmuxd.o
      ...
  "_plist_from_bin", referenced from:
      _create_plist_message in libusbmuxd.o
  "_plist_from_xml", referenced from:
      _receive_packet in libusbmuxd.o
      _create_plist_message in libusbmuxd.o
  "_plist_get_data_val", referenced from:
      _device_info_from_plist in libusbmuxd.o
      _usbmuxd_read_pair_record in libusbmuxd.o
  "_plist_get_node_type", referenced from:
      _usbmuxd_get_device_list in libusbmuxd.o
      _device_info_from_plist in libusbmuxd.o
      _receive_packet in libusbmuxd.o
      _usbmuxd_read_buid in libusbmuxd.o
      _usbmuxd_read_pair_record in libusbmuxd.o
      _create_plist_message in libusbmuxd.o
  "_plist_get_string_val", referenced from:
      _device_info_from_plist in libusbmuxd.o
      _receive_packet in libusbmuxd.o
      _usbmuxd_read_buid in libusbmuxd.o
      _create_plist_message in libusbmuxd.o
  "_plist_get_uint_val", referenced from:
      _device_info_from_plist in libusbmuxd.o
      _receive_packet in libusbmuxd.o
  "_plist_new_data", referenced from:
      _usbmuxd_save_pair_record_with_device_id in libusbmuxd.o
  "_plist_new_dict", referenced from:
      _create_plist_message in libusbmuxd.o
  "_plist_new_string", referenced from:
      _usbmuxd_read_pair_record in libusbmuxd.o
      _send_pair_record_packet in libusbmuxd.o
      _usbmuxd_delete_pair_record in libusbmuxd.o
      _create_plist_message in libusbmuxd.o
  "_plist_new_uint", referenced from:
      _usbmuxd_connect in libusbmuxd.o
      _send_pair_record_packet in libusbmuxd.o
      _create_plist_message in libusbmuxd.o
  "_plist_to_xml", referenced from:
      _device_monitor in libusbmuxd.o
      _usbmuxd_get_device_list in libusbmuxd.o
      _receive_packet in libusbmuxd.o
      _usbmuxd_connect in libusbmuxd.o
      _usbmuxd_read_buid in libusbmuxd.o
      _usbmuxd_read_pair_record in libusbmuxd.o
      _send_pair_record_packet in libusbmuxd.o
      ...
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libusbmuxd-2.0.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

```

jkcoxson, 2022-02-23:
You’ve installed for built the wrong architecture for libimobiledevice

mexmer, 2022-02-23:
your libplist is built for intel architekture, not for M1, you need to rebuild libplist first for M1, then you can build libusmuxd