libimobiledevice/libusbmuxd #131: libusbmuxd and libimobiledevice-glue built for Windows can lockup on a disconnect

sctol, 2023-08-18:
Ran into an issue using libimobiledevice tools locking up with a iOS device connected with a faulty USB cable and tracked it down to libusbmuxd and libimobiledevice-glue.

Code linked against the Windows MSVC runtime library using its send(), recv(), and select() functions can return on a hard disconnect or connection reset with errno set to 0.

One possible workaround would be to replace returns of -errno with (errno == 0) ? -ECONNRESET : -errno

libusbmuxd/src/libusbmuxd.c around lines 172 and 208
ibimobiledevice-glue/src/socket.c around line 1225

sctol, 2024-02-23:
Still running into lockup issues on Windows with send() in libusbmuxd and recv() in libimobiledevice-glue/socket.c

Specifically at lines 210 and 224 of libusbmuxd.c, suggest changing "res = -errno;" with "res = (errno == 0) ? -ECONNRESET : -errno;"

and at line 1595, change "num_sent = errno;" with "num_sent = (errno == 0) ? ECONNRESET : errno;"

nikias, 2024-02-23:
See my comment here:
https://github.com/libimobiledevice/libimobiledevice-glue/issues/41#issuecomment-1961868247
working on a solution.